CrazyDude1994/android-taranis-smartport-telemetry #4: Builtin bluetooth module support

froqstar, 2019-03-29:
Many of the current FrSky radios have a builtin bluetooth (LE) module, that can be used for wireless trainer or telemetry transmission (see for example the "FrSky Dashboard" app in the Play Store).

Is it possible to support this internal bluetooth module as well? 
That would be great, since then you wouldn't need any additional hardware on radios that support it (Q X7S, X-Lite Pro, X10, X12...)

leeph, 2019-03-29:
Seconded. I was just trying to find a way to contact you to ask the same thing.

My guess would be that the onboard Bluetooth telemetry in Horus X10S (for example) is outputting the plain (inverted) smartport signal, so to make this work would require the inversion in software on the app side....

Would this be something you could code? Seems like it ought to be fairly straightforward to code a method to invert the data from the Android device's Bluetooth before sending it through to the app proper for decode and display.

CrazyDude1994, 2019-03-29:
Hi guys. I'll take a look into it. One thing is I need someone to test changes.

leeph, 2019-03-29:
Happy to help you test, dude. I have Horus X10S, FYI.

CrazyDude1994, 2019-03-29:
Can you guys tell me what exactly happens what you try to connect?

CrazyDude1994, 2019-03-29:
Also that would be great if you could download https://play.google.com/store/apps/details?id=com.macdom.ble.blescanner&hl=en this app, connect to your module, and make a screenshot of the services and characteristics it provides.

leeph, 2019-03-29:
Have a look here.

https://youtu.be/T-VdLfcyatc

CrazyDude1994, 2019-03-29:
You can try this build 
[app-debug.apk.zip](https://github.com/CrazyDude1994/android-taranis-smartport-telemetry/files/3023810/app-debug.apk.zip)

leeph, 2019-03-29:
I've installed that. Behaviour is slightly different but ultimately fails to connect.

https://youtu.be/3JV9xyeM1QA

CrazyDude1994, 2019-03-29:
[app-debug.apk.zip](https://github.com/CrazyDude1994/android-taranis-smartport-telemetry/files/3023887/app-debug.apk.zip)
Try this one. If this one fails also, I would need to make a version with more debug info output

Codetron, 2019-03-31:
@CrazyDude
I managed to connect to the radio by setting it as a slave in the trainer section and using pin 000000 ... the radio indicates that the phone is connected but the app cannot connect ... I found a project that uses the internal bt module to play at the pc here is the link https://github.com/garyjohnson/opentx-ble-joystick/blob/master/README.md

great job 😃

if you need a tester I am available 😋 ...

my radio: Taranis Q X7S
open tx version: 2.2.3

CrazyDude1994, 2019-04-02:
If it's BLE, you should not pre connect to it, just try to connect with the app.

CrazyDude1994, 2019-04-04:
So in order to support this I need someone's help. I've made a test application which will log needed bluetooth data for me. What you need to do:

1. Download apk in this comment, install and launch
2. Launch your transmitter and your drone/wing so telemetry data is sending
3. Press connect BLE, it should show you your bluetooth device.
4. After this, it should show you the list of services your bluetooth provides. One of them should be a service that sends telemetry data. The only way to know is just to try all of them (there may be 1 or few). After service selected, it should show some data being sent if this the right one. If nothing happens, just disconnect and try another one.
5. If you successfully received some data, go to your sdcard/BluetoothLogs/ and there should be file named btlog.log, send it here with your transmitter model
6. Hopefully wait for me to implement it.

[bt_debug-debug.apk.zip](https://github.com/CrazyDude1994/android-taranis-smartport-telemetry/files/3045216/bt_debug-debug.apk.zip)

leeph, 2019-04-05:
Thank you for your efforts dude - I will run the test either this evening or tomorrow and update this post with the results.

leeph, 2019-04-13:
Sorry I didn't get around to doing this last weekend but it is on my to-do list for this weekend (possibly this evening). Will update shortly.

CrazyDude1994, 2019-04-14:
> 
> 
> Sorry I didn't get around to doing this last weekend but it is on my to-do list for this weekend (possibly this evening). Will update shortly.

No problem, take your time

maxchen2k, 2019-04-15:
[btlog.log](https://github.com/CrazyDude1994/android-taranis-smartport-telemetry/files/3079572/btlog.log)

My config as following:
1.Frsky X9D+ SE (OpenTX 2.2.3) 
2.R9M & R9MM ( last Flex version)
3.R9MM connect to F4 FC with inverted-fport
4.my connection please check pictures.

I hvae get some data from HC06 module in android. but iNav Telemetry view not work.

[image]
[image]
[image]
[image]

leeph, 2019-04-15:
That is not wired up correctly. You've got the middle leg of the transistor going to the Bluetooth 'TX' pin.

Your yellow wire should not be attached to the 'RX' in on the Bluetooth module at all. The yellow wire should be attached to the middle leg of the transistor, and the middle leg should not be connected to ANYTHING else.

maxchen2k, 2019-04-15:
would you please provide correct wiring picture ? thanks

leeph, 2019-04-15:
The photo Crazydude has on the front page is accurate, as is the wiring diagram. I have also told you what you need to do to fix it - it is just two changes to what you've already done.

If you cannot follow those instructions, I don't think we can help you.

leeph, 2019-04-15:
[image]

Here you go. Best one I have on my phone.

maxchen2k, 2019-04-15:
thank you very much. I will try modify my mistake and try again.

maxchen2k, 2019-04-15:
My wiring after correct mistake as following picture. 
[image]

iNav Telemetry still not work. enclosed log file from btlogger.

[btlog.log](https://github.com/CrazyDude1994/android-taranis-smartport-telemetry/files/3079843/btlog.log)

leeph, 2019-04-15:
Did you configure the HC-06 module with the correct settings? 57,600 baud (AT+BAUD7), AT+ENABLEIND0, etc?

Also you should be aware this only works with craft using SBUS/S.port - not with F.port.

If that all fails, I'm not sure I have anything else to offer except that I have one fully tested left that is available for sale.

https://www.ebay.co.uk/itm/163631569478

maxchen2k, 2019-04-15:
I did AT commands with HC06 as well.  Maybe the f.port cause something not work. I will try work with s-bus/s.port later. thanks for your help.

leeph, 2019-04-15:
There's no maybe about it. Craft configured with F.port DO NOT WORK with this solution. Yet. CrazyDude is looking in to how it can be achieved, but for now this is only for SBUS/S.port.

Codetron, 2019-04-15:
While waiting for the flight controller to arrive, I have a log from Frsky X7S (OpenTX 2.2.3).
Before making the log I set the internal Bluetooth as telemetry output from the openTX hardware tab. I attach you a photo.
During the test, the radio was not connected to any receiver. If necessary these days I can repeat the test using a receiver connected to INAV.
Thanks again for your support!!!! 

[image]

[btlog.log](https://github.com/CrazyDude1994/android-taranis-smartport-telemetry/files/3081390/btlog.log)

CrazyDude1994, 2019-04-17:
> While waiting for the flight controller to arrive, I have a log from Frsky X7S (OpenTX 2.2.3).
> Before making the log I set the internal Bluetooth as telemetry output from the openTX hardware tab. I [...]

At first glance looks like it's valid s.port telemetry, just without any data. Try it with receiver and inav, should work

nmaggioni, 2019-05-19:
Here are the logs of the second service published by my X10S' internal BT module. One is of an F.Port model and the other of an SBus+S.Port one.

+ [btlog_fport.log](https://github.com/CrazyDude1994/android-taranis-smartport-telemetry/files/3195120/btlog_fport.log)
+ [btlog_sbus_sport.log](https://github.com/CrazyDude1994/android-taranis-smartport-telemetry/files/3195121/btlog_sbus_sport.log)

CrazyDude1994, 2019-05-21:
Hi. I have made new experimental version. After choosing your bluetooth device, there may be a dialog where you should select your BLE service. If dialog is showed for you, simply select first one, if it doesn't work, disconnect and then select another one until it will work (if it will). If dialog is not showing for you, it means that there is only 1 BLE service, and app will automatically choose it

[app-debug.apk.zip](https://github.com/CrazyDude1994/android-taranis-smartport-telemetry/files/3204018/app-debug.apk.zip)

nmaggioni, 2019-05-21:
I can report that this works with the X10S/R9/F.Port combination. The trick with this module is, at least on my device, the association (even though it's BLE): the first pairing request appearing on the phone will be generic and once that has been accepted a second one will pop up. Accepting it will finally ask for the pin, which is not the one the radio shows but `000000` (six zeroes instead of four) (opentx/opentx#6440).

So far I've only been able to test this on the ground, in the coming weeks I'll verify its behavior with GPS fix and while actually flying.

CrazyDude1994, 2019-05-23:
[ble support.zip](https://github.com/CrazyDude1994/android-taranis-smartport-telemetry/files/3212252/ble.support.zip)

This is release candidate for BLE support. You can try this and tell me what you think and if it's working for you @froqstar @leeph @maxchen2k @nmaggioni

maxchen2k, 2019-05-23:
Good! Will run test asap. Still need inverter?

nmaggioni, 2019-05-23:
All bench tests passed with my hw :+1: :heavy_check_mark:

CrazyDude1994, 2019-05-23:
@maxchen2k yes

leeph, 2019-05-23:
Uhhh, hang on, where is an inverter supposed to attach if you're using the transmitter's built-in Bluetooth?

Are you saying we have to cable up the R9M smartport output to the transmitter's Smartport socket, with an inverter in between, for this to work?

CrazyDude1994, 2019-05-23:
@leeph oops. Yeah you're right. No inverter needed for this.
@maxchen2k

leeph, 2019-05-23:
Ah cool, thank goodness. I will be testing the release candidate code this evening and will report back. Thanks again for your efforts Sergey.

leeph, 2019-05-24:
Great stuff, I can confirm I was able to pair my Horus transmitter's BLE service and get the stable candidate release of this software to connect.

I need to do some more testing to ensure the connection process is smooth because a couple of times the connection dropped off and I was prompted to re-pair. It could get annoying if the pairing is not retained permanently.

I also had a strange issue whereby I had a huge long vertical track spanning the entire map/globe - it seemed to be thinking my quad had been in Antactica... this was on a Betaflight quad though, I didn't get around to trying it with iNav which is obviously the target platform for this.

maxchen2k, 2019-05-24:
> [ble support.zip](https://github.com/CrazyDude1994/android-taranis-smartport-telemetry/files/3212252/ble.support.zip)
> 
> This is release candidate for BLE support. You can try this and tell me what you think and if it's working for you @froqstar @leeph @maxchen2k @nmaggioni

Transmitter:  X9D + R9M Flex version +BLE (CC2541 HM-10) + Simply hand made Inverter on BLE board. 

Aircraft: Matek F405 Wing + R9MM(F.Port)

WORK WELL!

You did good job. thanks

CrazyDude1994, 2019-05-24:
@maxchen2k well, external BLE support was already implemented in 1.3

closed 2019-05-25 by CrazyDude1994

skorokithakis, 2019-11-06:
Sorry to post on a closed thread, but I'm trying to add an HC-05 Bluetooth module to my JumperTX's internal bay so I can use this.

Does anyone know how to get this to work? I've located the pins but the app isn't showing anything, even though it works fine with inverted external SmartPort. Does anyone know what baud and other settings I need?

leeph, 2019-11-06:
To be fair, all the instructions are in the main section of this Github, how to build the module etc.

Smartport is 57600bps so that's what your BT module must be set to in firmware.

skorokithakis, 2019-11-06:
I have built the module and everything works externally, I'm just not sure about how the relay is done to Bluetooth (whether the internal and external UART baud rates are the same, which you answered). Do I still need the inverter or different app settings? It would seem that I don't.

leeph, 2019-11-06:
The iNav telemetry viewer android app requires an uninverted Smartport serial link. When the Smartport data exits your transmitter or module, it is inverted (normal Smartport). The inverter that we build onto the Bluetooth module inverts the inversion - so to speak - giving an uninverted signal to the Bluetooth adapter which passes that uninverted serial link to the Android device.

I hope that clears things up.

redjacketKR, 2019-11-06:
> I have built the module and everything works externally, I'm just not sure about how the relay is done to Bluetooth (whether the internal and external UART baud rates are the same, which you answered). Do I still need the inverter or different app settings? It would seem that I don't, which is one difference between the internal and external modules.

Where did you put your HC-05? Do you know you can also install frsky BLE module in T16?

leeph, 2019-11-06:
Actually I use BT-06 modules, but the process is essentially the same. I mount them in a small 3D-printed case and stick it on the back of my Horus. The Horus has a BLE module which Crazydude has managed to get working but it's a bit flaky so I stick to using my external Bluetooth adapter which works flawlessly.
[image]

skorokithakis, 2019-11-06:
@redjacketKR I put it where @leeph did and it works fine, but I want to mount it internally so I am trying to install it in my T16. I know you can use the FrSky BLE module, but I have an HC-05 so I'm trying to see if I can just get that working instead. From what @leeph said, I need an inverter for the internal module too, but it's all 3.3V so I'm making one like that and I'll test.

@leeph, flaky in what way?

leeph, 2019-11-06:
Random telemetry pauses and disconnects, issues establishing the connection etc. Its just a lot more reliable with the full Bluetooth module rather than the low energy type. It's not like we need the low energy version particularly given the drains on the battery by the RC modules etc, it's all small beer.

skorokithakis, 2019-11-06:
@leeph Oh, I see, so it's not the app itself. Then if I can get the internal connection working with the HC-05, it'll be the best of both worlds, but I can't seem to get it working at all. Inverted, uninverted, whatever baud, etc, the external module works fine but the internal refuses to.

I am thinking that the radio communicates with the internal module in a different baud rate, the OpenTX sources say 115200 but I've set it to that and no dice (uninverted). It seems odd to me that the internal module would require inversion as well, I can't see any inversion MOSFETs on the [internal BT module](https://www.horusrc.com/en/frsky-transmitter-horus-x12s-bluetooth-module.html).

leeph, 2019-11-06:
Smartport is always inverted (unless you uninvert it yourself, as with the BT adapter here).

It doesn't matter where you take the Smartport serial data from, it's going to be inverted no matter what, until you uninvert it yourself.

Smartport is also always 57600. The control link is no doubt a higher baudrate especially for the internal module, but that is not the bit you're tapping into.

skorokithakis, 2019-11-06:
Ah, then I'm afraid I'm out of ideas. I've wired everything up correctly, connecting the RX pin on the HC-05 to the SmartPort output on the R9 shows telemetry fine, but connecting it to any of the pins in the internal BT module does nothing... Unfortunately I can't see what's being sent, but the app is definitely not recognizing any telemetry.

leeph, 2019-11-06:
Just a thought then, because I know CrazyDude has got the app to work with the Horus internal BLE module, and there is no additional hardware inverted required to make that work, perhaps the serial output from the internal module to the internal BT module is already uninverted. Try taking the inverter off your HC-05 and hooking up to the internal pins directly.

wilco1967, 2019-11-06:
I don't have a Jumper Tx, but a standard Taranis, so this may be different

But did you enable S-port forwarding to the serial port in the radio
settings ?
Without this, the build in serial port will not output any data  (at least
on Taranis).

skorokithakis, 2019-11-06:
@wilco1967 I think so, where is that setting? I just finished putting the module inside the R9, so I hope it wasn't a simple setting that I missed :(

redjacketKR, 2019-11-06:
> Ah, then I'm afraid I'm out of ideas. I've wired everything up correctly, connecting the RX pin on the HC-05 (through the inverter) to the SmartPort output on the R9 shows telemetry fine, but connecting it to any of the pins in the internal BT module does nothing... Unfortunately I can't see what's being sent, but the app is definitely not recognizing any telemetry.

You are doing exactly where I stopped. Just install "x12s bluetooth 4.0 module" $9 and HM-10 module  $3~5. Set with AT commands and  Install OTX 2.3.2 in your TX for BL telemetry. 

Easier and a lot better for now I guess. Don't waste your time. I admit that I learn a lot.

holoduke, 2019-11-07:
Can you give me the AT commands you used to connect the HM-10 to te x12s bluetooth?

skorokithakis, 2019-11-07:
@redjacketKR That's easier, sure, but I want to find out what the problem is. If the radio is sending telemetry on the internal TX pin, why can't the HC-06 read it? I should have used my logic analyzer before I closed it all up.

wilco1967, 2019-11-07:
> @wilco1967 I think so, where is that setting? I just finished putting the module inside the R9, so I hope it wasn't a simple setting that I missed :(

The setting is in hardware menu, Serial Port -> Telem Mirror.

skorokithakis, 2019-11-07:
I don't have that setting on my Jumper T16, I'm afraid. Just "Bluetooth: Telemetry".

ZeekWorld, 2019-11-27:
@leeph Do you mind sharing the 3d print stl for your bluetooth module box? Thanks in advance.

[22 more replies]