CrazyDude1994/android-taranis-smartport-telemetry #5: Crossfire support

CrazyDude1994, 2019-03-29:


Saxin, 2019-06-17:
Hi any chance you know status of CF support ? I was under impression that internally on OpenTX all telemetry data is converted to SmartPort types and this should work out of the box ?
Anyway if I can help with something - please let me know ?

Saxin, 2019-06-17:
@CrazyDude1994  thanks a lot for the fast response, the situation as follows:

1. I use R9 and Crossfire micro TX - this one doesn't has built in bluetooth.
2. I've built inverter + HC-05 module and connected internally to X9D (Taranis) battery compartment serial port.
3. In Taranis port is set to S.Port Mirror, I have OpenTX 2.2 installed.
4. I've checked that the hardware side of things works perfectly when I'm connected via. R9 module to one of my planes, I get everything showing and working niocely.

5. Now I switch module to Crossfire micro and connect to another plane, on the plane using CRSF protocol, FC runs Inav - nothing comes out to app.

I would gladly help in any way possible, I can sniff data if needed have logic analyser, I have a feeling that this is more related to OpenTx side of things.

Saxin, 2019-06-17:
Found this one it seems related - https://github.com/opentx/opentx/issues/5357?fbclid=IwAR38XwTD6PsjfZAYJG36WxJRnewgWipy65oye_ZcO9i0hQWZP1KSh_TBSpg,

On my side, updated X9D to 2.2.4N703 (Nighty).
Tried both 56700 and 115200 (as this was mentioned in the issue) bauds without any luck with the app.

Next what ?

CrazyDude1994, 2019-06-18:
Looking at the log it looks like different protocol (which is no suprise). Now I need to find some protocol specification

CrazyDude1994, 2019-06-18:
Also @Saxin do you have telegram? If yes, join our telegram group (link in readme). That would boost implementation

Saxin, 2019-06-18:
@CrazyDude1994  Thanks for taking a look, it's crossfire native protocol, let me gather for you some sources I've looked at yesterday using it, 

1. RavenLRS (made by few Inav devs) https://github.com/RavenLRS/raven , 
Relevant files: crsf.c / crsf.h 
https://github.com/RavenLRS/raven/tree/master/main/protocols

2. OpenTX https://github.com/opentx/opentx/tree/2.3/radio/src/telemetry
crossfire.cpp / crossfire.h

3. Inav https://github.com/iNavFlight/inav/tree/master/src/main/telemetry
crsf.c / crsf.h

4. Betaflight https://github.com/betaflight/betaflight/tree/master/src/main/telemetry
crsf.c / crsf.h

I recall seeing somewhere protocol specification pdf - will take additional look for it.

CrazyDude1994, 2019-07-13:
Hi guys. Crossfire is almost there, but I still need your help.

I need a good data log in order to make it complete. I need a full flight log (wait for GPS lock, arm, fly, land). Instructions on how to record it can be found here https://github.com/CrazyDude1994/android-taranis-smartport-telemetry/issues/4#issuecomment-480041038

vquintans, 2019-07-15:
@Saxin Added some logs here: https://github.com/raul-ortega/u360gts/issues/47#issuecomment-511428329 Maybe they could be useful for you .

Saxin, 2019-07-15:
@vquintans  I've sent Sergey the logs via. telegram, he has them now I got some homework in return

CrazyDude1994, 2019-07-16:
If someone interested in trying test build, grab this apk and tell me the results:

[app-debug.apk.zip](https://github.com/CrazyDude1994/android-taranis-smartport-telemetry/files/3398462/app-debug.apk.zip)

vquintans, 2019-07-21:
Tried to test de test build without success...

I press "Connect" and select the bluetooth adapter. After a few secons a message apears on the bottom saying "Connection failed".

I attach a log captured with your debug app where it seems that the adapter and port mirroring is working: 
[btlog.log](https://github.com/CrazyDude1994/android-taranis-smartport-telemetry/files/3414719/btlog.log)
.

CrazyDude1994, 2019-07-22:
Hi guys, please try this new version: 
[app-debug.apk.zip](https://github.com/CrazyDude1994/android-taranis-smartport-telemetry/files/3418996/app-debug.apk.zip)

CrazyDude1994, 2019-07-22:
[app-debug.apk.zip](https://github.com/CrazyDude1994/android-taranis-smartport-telemetry/files/3419082/app-debug.apk.zip)
And here is a new version with new fix

putimir, 2019-08-19:
Hey @CrazyDude1994 : I'm trying this latest version with my Horus X10S and a Crossfire micro TX. I'm a bit overwhelmed with all the info, can you tell me straight if this should work? 

I'm getting telemetry via Crossfire link and works perfect...I would like to pass it to BT

Thanks

CrazyDude1994, 2019-08-20:
It should work only with an external bluetooth module connected to your transmitter. It wont work with the bluetooth in your Crossfire (if you have one) You need OpenTX 2.2.4 and set it to mirror telemetry data

putimir, 2019-08-20:
Hey, we're talking about TBS Micro TX, and a Horus X10 w/internal/external BLE device; so, if undestand correctly, the TX already parses CRSF TLM data from TBS module through serial link (JR bar), and then pushes that data, supposedly s.port formatted, to the internal/onboard BLE device...

Saxin, 2019-08-20:
@putimir - yes OpenTX parses the CRSF and just puts it "as is" to the aux serial port hence the word mirroring, it's not s.port formatted it's native CRSF telemetry data. (the term S. port mirror in Taranis menu is confusing and better changed to something like UART mirror IMO)

putimir, 2019-08-20:
Hey @Saxin: is the Horus internal BLE an "aux serial", when it's set to "telemetry" ?

jasc76, 2019-10-05:
Hi
got it to connect to the internal BT of my qx7s (crossfire module) to an INAV wing.
But connection drops every minute or so

CrazyDude1994, 2019-11-08:
Here is a beta version. Please check it out and give me your feedback

https://play.google.com/apps/testing/crazydude.com.telemetry

afernan, 2020-02-19:
Hi:
I have a crazy dude:
The app you did (thanks a lot!) works very well with my X9D-2019 SE using the internal BLE of the radio.
I´m using CRSF system (nano Rx, and Micro module and also tested with STD module). 
I guess the "mirror telemetry" in the radio was using CSFR Telemetry (the INAV Flight Controller emiter). 

My doubt: the Telemetry obtained in the BLE is "CSFR" telemetry? or its S.Port Telemetry ?

I want to get the GPS from that BLE out for an antenna tracker that I´m doing.

CrazyDude1994, 2020-02-19:
@afernan I can't tell for sure.
You can send me an log file, and I will tell you which protocol it uses. You can send just a little piece of it

afernan, 2020-02-20:
@Sergey:
thanks a lot!
Attached a LOG.
The setup is : 
- an Omnibus F4V5 fc with GPS with INAV 2.4.0. Telemetry set as "SmartPort 57600baud"
- Receiver: TBS Nano 
- Transmiter: Taranis X9D SE 2019 (with BLE) with TBS Crossfire Micro. Confifured Bluetooth = Telemetry

Please let me know if you need anything else.
Thanks again

[btlog.log](https://github.com/CrazyDude1994/android-taranis-smartport-telemetry/files/4231036/btlog.log)
[btlog-HEX.log](https://github.com/CrazyDude1994/android-taranis-smartport-telemetry/files/4231037/btlog-HEX.log)

afernan, 2020-02-20:
btw : this the antenna tracker I´ve build. It works fine based in reading the TBS Csfr module Bluetooth (using Mavlink) but I want to move to the Taranis internal BLE that is more universal.
[image]
[image]

afernan, 2020-02-21:
This other LOG has been recorded using the app "Telemetry viewer"

[2020-02-01 10-19-06.zip](https://github.com/CrazyDude1994/android-taranis-smartport-telemetry/files/4235535/2020-02-01.10-19-06.zip)

CrazyDude1994, 2020-02-21:
@afernan it has crossfire data

closed 2020-10-09 by CrazyDude1994

MartinGiersberg, 2023-06-10:
> btw : this the antenna tracker I´ve build. It works fine based in reading the TBS Csfr module Bluetooth (using Mavlink) but I want to move to the Taranis internal BLE that is more universal. [image] [image]

Did you manage to connect the bluetooth transmitter in your Radiocontroller to the bluetooth reciver in your arduino ? Im currently working on a similar project and im unable to make that connection happen