CrazyDude1994/android-taranis-smartport-telemetry #10: For electronic noobs, please describe in words how to wire up the HC-06 & inverter bits

leeph, 2019-03-30:
Hi Crazydude,

I'm an electronics noob and I don't fully understand the various symbols in the diagram for hooking up the inverter.

Please could I ask you do describe in words below the diagram which legs of the MOSFET go to which other bits. I've looked at both the photo and the diagram and I'm still slightly confused so just a paragraph to say what goes where would be very useful!!

wilco1967, 2019-03-30:
slightly different scheme, using slightly different parts,  but works just the same.

https://github.com/stronnag/mwptools/wiki/Building-a-bluetooth-adapter-for-Taranis-S-port-telemetry-for-MWP'

https://drive.google.com/file/d/0BwevcPezRywJaVdjQzZSdXJXcE92Z3BvelNXOXVfc0xTUGxv/view

leeph, 2019-03-30:
OK I've put the thing together, phone connects to the module OK but I get no telemetry in the app at all. I measure basically no voltage between ground and the R9M snartport pin on the back... is that right?

BT module was set to 57600, and the other commands too. It kind of looks all right but now I'm stumped. Can I send you a photo so you can sanity check my work?

For all I know the resistors I was sent might not be the right ones. I've definitely got a 2N7000 mosfet tho. Might I have damaged it while soldering it or something?

leeph, 2019-03-30:
https://drive.google.com/file/d/1PtAs8uvuRv1J2JIGPLum0iR0eIT9mP3d/view?usp=drivesdk

leeph, 2019-03-30:
Ok... turns out it does work, when I tried it on my Harrier which has R9Mini connected with s.port and SBUS. The first craft I tried it with was my Mini AR Wing which is connected by f.port.

So... it doesn't work with f.port, basically. I guess that's obvious when you think about it but I kind of assumed smartport would still be pushed out the back of the R9M even if the receiver itself is f.port connected. 

Might wanna update your page to highlight that caveat!!

CrazyDude1994, 2019-03-30:
That is odd. Can you discover sensors on your TX? Can you edit any sensor and make a photo. Also photo of the sensors will be great.

leeph, 2019-03-30:
Yes, like I said telemetry to the TX is working fine with my f.port craft. It looks just like smartport telemetry on the TX. No difference at all. The main protocol difference though is that f. port operates at 115,200bps natively, which is one major reason it's not working since the BT module is configured for 57600 for s.port. If I hadn't already got rid of the header pins on the BT module then I'd change the module port speed to 115,200 to test whether that would fix it.

I do have another BT module coming in the post so even that arrives I'll do some more testing with that and f.port.

leeph, 2019-03-30:
Does your app force the device's bluetooth serial port to 57600? if it does then i guess f.port definitely isn't going to work even if I change the transmitting module port speed.

CrazyDude1994, 2019-03-30:
You don't "force" or choose "baud rate" for wireless bluetooth communication (atleast in Android). It will work whatever it is

P-I-Engineer, 2019-04-01:
these are the inverters i'm using
https://www.aliexpress.com/item/2pcs-No-Welded-Naze32-Cleanflight-SBus-Signal-Inverter-Connect-Cable-Confirmation-Signal-for-Naze32-Flight-Control/32720822861.html?spm=a2g0s.9042311.0.0.595c4c4dZLhYUv

leeph, 2019-04-01:
What does the ENABLEIND0 command do exactly, and is it an absolute requirement for the BT HC06 module to work for this? I have one unit with different firmware than the others and it's not a supported command it seems. Consequently this particular one doesn't display any telemetry in the app so just want to check for sure this is the reason why?

CrazyDude1994, 2019-04-02:
It makes so BT does not send own data (status changes, errors, etc). Otherwise it may corrupt actual data

Ellou, 2019-04-05:
Anyway around using ENABLEIND0. I started another issue for my hc 05 not recognizing that command. Haven't tried it out yet but hopefully will this weekend.

leeph, 2019-04-05:
I've got HC06s that don't even have that command. There seems to be a large number of different firmware builds for all these bluetooth boards and even the AT command syntax varies between them.

From now on i'm buying only BT-06 which seems to be the latest iteration of hc-06 and definitely supports the enableind command.

Ellou, 2019-04-05:
Do the boards without that command work?

P-I-Engineer, 2019-04-05:
Mine does

leeph, 2019-04-05:
Mine didn't. Couldn't fathom any other reason why. The 2 I've built with BT-06s both worked great. The HC-06 missing the command didn't receive any telem data.

Ellou, 2019-04-05:
Uuggghhhh. Now I have to order another one.

P-I-Engineer, 2019-04-05:
Hc-06 and hc-05 both working for me.

Ellou, 2019-04-05:
Did you use the ENABLEIND0 on both?

P-I-Engineer, 2019-04-05:
Neither

Ellou, 2019-04-06:
Just wanted to update. My module connects and I am able to see telemetry data in the app. Only issue is it stays connected for about 30 seconds and the app freezes. I did also get sensor lost so I think that may be the issue. On to troubleshoot that issue. Thanks for creating this and all the help.

Ellou, 2019-04-06:
So I tested further and seems like my samsung note8 edge will not stay connected. My old nexus 5 or 6 stays connected without an issue. Anyone else have an issue like that? I'll have to check another device to see if it's just the one phone. 

Also is everyone else's pretty laggy? Mine updates every 5 to 10 seconds sometimes. Not sure if that's the best we get or not. 

Sorry if this isnt really the place for this. Is there a forum or thread for this somewhere?

leeph, 2019-04-09:
Rather than open another thread just to ask this question, Crazydude could you help me understand why the hardware inverter is required at all?

If the Smartport data was sent natively (inverted) to the Bluetooth module (and thus to the device running the telemetry app) would it not have been possible to do the logic inversion as part of the app's code? Negating the need for the inverter, making it work with stock unmodified Bluetooth adapters?

If not possible, please help me understand why, as it's a question I've been wondering about in relation to another project i'm working on.

Cheers, leeph

CrazyDude1994, 2019-04-10:
`it not have been possible to do the logic inversion as part of the app's code`
No, it's not possible. It simply comes out as a noise. If that were possible, there were no problems in RC community using S.port or sbus

leeph, 2019-04-10:
Well, I see what you are saying, but isn't that exactly what happens in the case of a softserial port? The inversion is done in software so that a hardware logic inverter isn't necessary.

It takes CPU cycles so not suitable for low latency serial links, but it is perfectly acceptable for telemetry, so can you not pull the code for softserial from inav/betaflight Github and port to your android app?

KFNCOLO, 2019-04-11:
This might not be the best spot or thread (and maybe it's even common knowledge), but I landed here after days of searching and trying to figure out how to configure my HC-06, so maybe it will help others. I'm using the Arduino IDE for this configuration, which has worked in the past on other HC-06 BT boards. 

The only AT command that worked was AT, and AT+VERSION and only using "both NL & CR". The version of my HC-06 is VERSION:3.0-20170609, versus Livignor or something like that, which I guess is more common. I finally found that commands are different in this version. Here are a few commands that I have found that work.
Change name: AT+NAME:DesiredName
Change baud rate to 57600: (or whatever rate needed): AT+UART=57600,0,0
Find out Baud rate: AT+UART?
Change pin: AT+PSWD:"0000" 
Related to some above posts though, I have not been able to figure out the AT+ENABLEIND0 to get it to work. 
So, I will go ahead and finally solder this up and see if it works without that change.
@CrazyDude1994 Thanks so much for coming up with this app, something I've been trying to find for awhile now. Very appreciated.

pz101, 2019-04-13:
Yes ... thank you Crazy Dude. An app of this caibre has been long awaited. I have a drawer full of HC-05's. I've added mosfet inverter and resistor. And my android connects no problem. Your app connects as well but no data. I'm using an R9M and both FRSKY GPS and Lipo telemetry sensors. All telemetry gets to my Q X7S but bad or no data (not sure which) to the app from s.port on the R9M. I noticed everyone using HC-06. Is the HC-05 a bad idea? Also none of my HC-05's have AT+ENABLE0 All report error(0). Any suggestions would be greatly appreciated.

CrazyDude1994, 2019-04-13:
> Yes ... thank you Crazy Dude. An app of this caibre has been long awaited. I have a drawer full of HC-05's. I've added mosfet inverter and resistor. And my android connects no problem. Your app [...]

Are you using Inav? Because this app only supports Inav sensors, not the original that Frsky has

DiscoMan18, 2019-04-19:
Would be also VERY VERY nice from your side to add the support of original Frsky sensors.

pz101, 2019-05-02:
Ah ha.... Thank you for clearing this up. Great App... I wish I could use it with frsky sensors. Oh well. I'll move on to the next project. Keep up the good work CrazyDude.

closed 2019-06-05 by CrazyDude1994