CrazyDude1994/android-taranis-smartport-telemetry #14: F.Port to bluetooth

tasi0000, 2019-04-06:
OK, following up to another issue (not tested there), I can confirm that F.Port protocol is a no go with the recommended setup configuration. Setup is a BFF3 board and R9mm receiver on F.Port on UART TX1 with working telemetry.
What I did--- configured bluetooth HC-05 (no button) and changed the baudrate to 115200 to conform to the F.Port baud. I could not pass the ENABLEIND0 command though. All this with a USB to serial with breakout via arduino. 
Nevertheless, all the rest, wiring programming etc. is fine. 
Secondly, I also tried to connect the telemetry RX wire from the MOSFET to the module TX and thirdly the taranis activated port mirror. No go. 

Although the app connects normally to the bluetooth, no information is passed to the app. On the Taranis telemetry LUA screen all is normally shown. 

I think we'll need a 115200 version of the app configured to work with F.Port to have this to work. 
@CrazyDude1994 any feedback or thoughts on the above?
@leeph, I guess you wont have to wait anymore for these modules to arrive 😄

P-I-Engineer, 2019-04-06:
why would f.port matter? that's just the actual rx side. You don't have to do anything to your radio to use f.port on your receiver. I would assume that it's still regular s.port that the radio is reading.

did you use an inverter?

tasi0000, 2019-04-06:
No I didn't use an inverter.but since both Rx and TX are flashed with F.Port protocol, it can't be the same as S.port. they also work on different baud rates. I'm not sure I understand what you want to convey. Do you have it working?

leeph, 2019-04-06:
I've tried the BT-06 module I built (with your inverter wiring) - it works brilliantly with all my s.port craft, but no telemetry is received in the app with the f.port craft.

Like I said, I have a feeling it's just the port speed. For example serial passthru updating of FrSky receivers doesn't work when craft is configured for f. port, because the port speed is 115200 where the frsky update tool wants 57600. You have to set the board back to s.port config temporarily to force the speed vack to 57600 for the update then switch back.

P-I-Engineer, 2019-04-06:
what tx are you using, i've used f.port in the past and did nothing to my transmitter.

tasi0000, 2019-04-06:
@P-I-Engineer using a taranis x9d. Everything is working as I said above, apart the data exchange from R9M to Bluetooth to app.

leeph, 2019-04-06:
Horus X10S here.

Same as above - the only thing different in the two scenarios is one craft is f.port and one is s.port. Everything else is identical.

CrazyDude1994, 2019-04-08:
F.port seems like another protocol than s.port. What port do you use to connect to your bluetooth? Do you connect it directly to R9M, or do you connect it to your transmitter with S.port mirror enabled?

tasi0000, 2019-04-08:
I tried both ways with no success unfortunately.

dangerg, 2019-04-09:
Try the same wiring with sbus. Make sure its not your inverter.

leeph, 2019-04-09:
Dude we've both done that.

We've both built a bluetooth module with the inverter. We power up a craft running SBUS & S.port - telemetry. We power up a craft running F.port configuration - no telemetry. I don't know how much clearer I can be on this subject. It's not the hardware.

CrazyDude1994, 2019-04-09:
The thing is that there no much info. I already told in another ticket what you need to do in order to help. That thing will help with this too

tasi0000, 2019-04-09:
@CrazyDude1994 can you link the other ticket? Scrolled over and couldn't find anything that relates to this one. What kind of info would you need? 
A thought that I'm having now in relation to the inverter is; could it be that the r9mm F.Port wont need inversion at the bluetooth? Since the convention is backwards that would make the pin named "Inverted S.Port" uninverted and "S.Port/F.Port" inverted; common misunderstanding with this receiver. Although this works with the flight controller and radio to understand the telemetry, inverting the signal at the bluetooth again might create the bottleneck.....
@leeph I can't take the inverter off right now to test, but have you tried it like this i.e. no inversion at the bluetooth?
@CrazyDude1994 does that make any sense, or am I totally and utterly wrong here?

CrazyDude1994, 2019-04-09:
https://github.com/CrazyDude1994/android-taranis-smartport-telemetry/issues/4#issuecomment-480041038

dangerg, 2019-04-11:
I converted one of my planes from fport to sbus + smartport telemetry.
I can confirm same behaviour. Works right away with smartport but nothing on fport.

CrazyDude1994, 2019-04-11:
As I already said, I cannot do anything about it without additional info.

CrazyDude1994, 2019-04-19:
Okay I have spare FC which I configured to work as Fport. Will be trying to decode some telemetry data...

CrazyDude1994, 2019-04-20:
[app-debug.apk.zip](https://github.com/CrazyDude1994/android-taranis-smartport-telemetry/files/3099604/app-debug.apk.zip)
I managed to make it work with fport. Just want to point out that every step is still required to do in order to make it work (set baud 57600, connect with inverter)

leeph, 2019-04-20:
Confirmed, I get telemetry through to the app with this version on my f.port craft too. Well done CrazyDude!

What changes did you have to make to the code to make this work, out of interest?

tasi0000, 2019-04-20:
That's great news!
What about the control? FPort needs to work on a higher speed 115200. Any adverse effects on control or telemetry and did you change both the port speed in configurator and Bluetooth module?

leeph, 2019-04-20:
I haven't touched the BT module port speed so it's still running at 57600 - which indicates the telemetry stream on f.port is still 57600 even if the control link is 115200. I'm interested to know what code changes in the app were needed to make it work.

CrazyDude1994, 2019-04-20:
As I said already, you don't choose any baud rate when using bluetooth. It is used between wire link. R9M uses 57600, taranis back port outputs 57600 too, so you need to configure BT to work on 57600.  There was literally one line of code being changed. For fport it just have different Hardware Id, so I just removed check for it which will allow to work on any hardware id

leeph, 2019-04-20:
Thanks for the info. I assume you will be wrapping f. port support into the the next version released on Google Play?

Also I did try to run the Bluetooth debug on the Horus Bluetooth integrated telemetry but because Android fails to pair the device it is impossible to get the debug app to display anything useful. If you try Bluetooth Low Energy the app just crashes.

I don't know whether it's worth persuing this avenue. It would be a nice to have but I'm not sure it's achievable when you can't even pair the Bluetooth device to start with.

CrazyDude1994, 2019-04-22:
I have released 1.3 to google play.

tasi0000, 2019-04-22:
@crazydude1994
Used your apk above and it's working!

dangerg, 2019-04-22:
Thanks crazydude

closed 2019-04-22 by CrazyDude1994