CrazyDude1994/android-taranis-smartport-telemetry #15: mAh in the battery icon are displayed as %

puspis, 2019-04-06:
First of all, very good work.
On the app screen the milliamperes consumed (mAh) are shown with the suffix "%"

In a new version, it would be possible to customize the telemetry sensors Alt, GAlt, Hdg, etc. and also customize the number of decimals for some sensors such as VFAS, Alt or Distance (for example 10.3km instead of 10372m)

Thank you in advance!

CrazyDude1994, 2019-04-06:
Yeah I know about this issue. The thing is that user can set any `smartport_fuel_unit` CLI. So I guess I need to add also setting in the app where you will select your unit.

puspis, 2019-04-15:
Hello. I changed the CLI parameter you said to PERCENT and now, the screen shows the correct value.

CrazyDude1994, 2019-04-16:
Fixed in 1.2. You can now select which units are you using

closed 2019-04-16 by CrazyDude1994

puspis, 2019-04-16:
Ok, thank you. I'll check

DiscoMan18, 2019-05-22:
Version 1.3. iNav 2.1.0.

smartport_fuel_units set to mAh. In the settings of the app Battery units are set to mAh as well, but the app still shows battery level as %. Milliamperhours visible in OSD are increasing, battery level stays shown in app stays the same. 

Seems like the issue is still not fixed... :(