CrazyDude1994/android-taranis-smartport-telemetry #17: [Support] QX7 + jdy-08 BLE module = no telemetry in app

DiscoMan18, 2019-04-09:
1. Quadcopter with iNav 1.9.0, Telemetry receiver R-XSR. 
2. Taranis Q X7 sees RSSI and receiver VBat voltage (I haven't connected R-XSR s.port to iNav yet) and supposedly should also be able to transmit it to app via bluetooth
3. jdy-08 configured via at commands, I set the name and baud rate to 57600.
4. wired BLE module to taranis bay module, powered it via bec (cause I use 2S pack to power my x7) and connected rx pin of BLE module to s.port pin
5. opened the app and successfully connected to my ble module. 
6. But no telemetry data within the app, I expected to see rssi and voltage

What I did wrong?

CrazyDude1994, 2019-04-09:
`I haven't connected R-XSR s.port to iNav yet` this
This data isn't transmitted via s.port (at least from the back of the R9M)
So you need to connect s.port to inav. Also, currently RSSI is not supported also because the reason it's not transmitted (as I said at least using the back of the R9M)

DiscoMan18, 2019-04-09:
No, I'm not using R9M. I'm taking s.port from bay module connector of my X7. Yesterday I forgot to use an inverter circuit for s.port and have wired ble module without it. Today I corrected this, but still no luck. Ok, next I'll try to feed telemetry directly from iNav to my R-XSR as you suggested.

DiscoMan18, 2019-04-10:
Now I can confirm, that with s.port of R-XSR connected to iNav everything works just fine.

closed 2019-04-10 by DiscoMan18