CrazyDude1994/android-taranis-smartport-telemetry #19: QLRS support

DiscoMan18, 2019-04-10:
Hello Sergey,

I'm writing in English as the question could be potentially interesting for English speaking auditory, but, if you like, we can continue in Russian. 

Yesterday I managed to feed the telemetry from iNav to the app via R-XSR rx, Taranis Q X7 and jdy-08 BLE module. At this point everything is working ok. The next step is to use your app together with iNav but via QLRS (qczek.beyondrc.com) which first encapsulates telemetry data from iNav in mavlink format, transmits it via air (using LoRa modems on which it's based) and on TXs side (i.e on the ground) converts received telemetry to frsky smart port format. Then the telemtry is sent to Taranis by R-XSR which acts as a relay for both SBUS and smart port data. Taranis sees received telemetry, but your app doesn't. I think the problem is your app expects to see somehow different ID's (or names) of the sensors in telemetry feed. Here is what is sent to Taranis by QLRS:

APID_GPS_COURSE 0x0840
APID_RSSI 0xF101
APID_VFAS 0x0210
APID_CURRENT 0x0200
APID_CELLS 0x0300
APID_ALTITUDE 0x0100
APID_VARIO 0x0110
APID_GPS_SPPED 0x0830
APID_LATLONG 0x0800
APID_GPS_ALTITUDE 0x0820
APID_AIR_SPEED 0x0a00
APID_FUEL 0x0600
APID_T1 0x0400
APID_T2 0x0410
APID_PITCH 0x0430
APID_ROLL 0x0440
APID_MAV_BASE_MODE 0x04A0
APID_MAV_SYS_STATUS 0x04A1
APID_MAV_CUSTOM_MODE 0x04A2
APID_CUST_RSSI 0x04B0
APID_RX_RSSI_REG_VAL 0x04B1
APID_RX_SNR_REG_VAL 0x04B2
APID_RX_PACKET_LOST_VAL 0x04B3

would it be possible to forward this data to you app? 

Regards,
Alexander.

DiscoMan18, 2019-04-10:
I think it may also be useful to allow a user to choose which telemetry values to display on screen and also to let him choose a correct Id.

CrazyDude1994, 2019-04-10:
I think there a place for it in the feature. I can't say it will be priority for now. There are much more important things to do first. As a feature to let user choose sensors, this is already on my plans.

DiscoMan18, 2019-05-14:
I'm closing the issue as everything works perfectly on version 1.3 ;) Tested on two different inav version: 1.9.1 and 2.1. Everything is working, the app shows all important data, including (the most important thing!) the position of the plane on the map. Thanks and hope to see your great app developing and getting only better!

closed 2019-05-14 by DiscoMan18