CrazyDude1994/android-taranis-smartport-telemetry #21: Slow update rate

CrazyDude1994, 2019-04-14:
Several people reported that telemetry update is too slow. It works pretty fast for me, so I want to figure out whether it app's fault or hardware thing.

You can help me by posting here your hardware, android phone model/version, and your update rate. That may help me somehow

puspis, 2019-04-15:
Hello I thing there is a delay of 1-2s. Not bad for me.
Taranis X9D + R9M. Bluetooth HC-06 + BS170 MOSFET
Smartphone is Xiaomi Redmi Note 4 running Android 7.0

CrazyDude1994, 2019-04-15:
Hi @puspis 
Thanks for the info, on question tho: are you talking about the delay or data update rate? Because I'm interested in the update rate (how much times it update in one second)

puspis, 2019-04-15:
The update rate is about 1.3 frames per second. But, all sensors do not refresh at same time.
Some measures:
Alt: 1.34 fps
mAh: 1.34 fps
Dist: 1.33 fps
VFAS: 1.36 fps
Regards

CrazyDude1994, 2019-04-15:
@puspis do you mind to test this apk and tell me the results of update rate? Don't mind the horizon view, it's still WIP.
[app-debug.apk.zip](https://github.com/CrazyDude1994/android-taranis-smartport-telemetry/files/3082444/app-debug.apk.zip)

CrazyDude1994, 2019-04-16:
I have released new version 1.2 on google play. Please tell me if you have any difference

DiscoMan18, 2019-04-19:
1-1.5 sec update rate. jdy-08 BLE module + Xiaomi Redmi 5 Plus + Taranis Q X7

CrazyDude1994, 2019-04-20:
[app-debug.apk.zip](https://github.com/CrazyDude1994/android-taranis-smartport-telemetry/files/3099603/app-debug.apk.zip)
You can try this one, it should work much faster

CrazyDude1994, 2019-04-22:
Released 1.3 to google play
You can check it here

closed 2019-05-21 by CrazyDude1994