CrazyDude1994/android-taranis-smartport-telemetry #24: Add support for LTM telemetry

giacomo892, 2019-05-01:
as for the title, should be great to have.

CrazyDude1994, 2019-05-01:
@giacomo892 I'm trying to understand how this is often used? Do I need seperate hardware to receive and transmit LTM data? Also how can I configure Inav to work with it?

giacomo892, 2019-05-01:
You can check LTM data format into INAV source code. It is very simple. INAV will output LTM telemetry on any UARTs you want and the signal is NOT inverted.
It's used a lot in trackers and monitoring ground stations like MWP so it would be nice to display and log position and other data on the phone with your app receiving data via bluetooth

CrazyDude1994, 2019-11-08:
@giacomo892 here is a beta version with a LTM support

giacomo892, 2019-11-09:
@CrazyDude1994 thanks! will try it out ASAP!

CrazyDude1994, 2019-11-09:
@giacomo892 I have also added usb serial connection. I'm closing this issue, please provide your feedback at https://github.com/CrazyDude1994/android-taranis-smartport-telemetry/issues/47

Thanks!

closed 2019-11-09 by CrazyDude1994