CrazyDude1994/android-taranis-smartport-telemetry #25: Customize sensors on screen

puspis, 2019-05-02:
Hello.
It would be possible to customize the telemetry sensors Alt, GAlt, Hdg, etc. shown on screen and also set up the number of decimals for some sensors such as VFAS, Alt or Distance (for example 10.3km instead of 10372m)
Thanks

CrazyDude1994, 2019-05-04:
That's what I'm working on right now. I think make it like opentx sensor scanner thing

DiscoMan18, 2019-05-14:
And also would be very nice to display current GPS coordinates of the plane along with other telemetry data. And in case if the telemetry data stream suddenly stops (due to crashed plane, for example) maintain last good gps coordinates on the screen. This is very useful when you lost your plane.

CrazyDude1994, 2019-05-14:
@DiscoMan18 for this purpose there is Replay log function. You can just use this to find latest known location

DiscoMan18, 2019-05-14:
Yes, but as far as I know it's only shown as a position on the map. And I'm talking about having the last good GPS coords in numeric form, just to be able to put them into Google Maps during search and rescue mission ;)