CrazyDude1994/android-taranis-smartport-telemetry #26: Feature request - periodic vocal alerts

leeph, 2019-05-04:
Please could I suggest that you implement configurable periodic spoken alerts for telemetry data values? Similar to what is available in iNav Mission Planner for android, but being configurable in terms of what values are spoken and at what rate /frequency.

CrazyDude1994, 2019-05-12:
That will be added on some stage

rsilk1949, 2020-02-02:
This can be handled on your radio (OpenTX, erskyTx, etc.) by using "voice alerts". I don't see any need to duplicate this in the app.

leeph, 2020-02-02:
The reason was because there is no ability to route the audio from the TX voice audio outputs to anything other than the tinny transmitter speaker.

If the app were able to generate spoken alerts using onboard Android voice synthesis, this can be routed via Bluetooth or wired audio to a car speaker system or Bluetooth speaker or whatever else.

That was the reasoning behind the feature request.