CrazyDude1994/android-taranis-smartport-telemetry #27: Add support for using Google map in China

rivermtnn, 2019-05-06:
Hi Dude, thank you very much for sharing this tremendously useful program. I could feel more comfortable to fly out a bit far with INAV with your work. I've used R9M with HC6 module to things working smoothly.

Just one issue for me particular maybe, that's I'm living in China with the normal Google map service is blocked by a firewall. I could make the app to access the map service by other means, however the plane location is dislocated on the map view. Because there is some human induce interference over the data from map service.

There is a dedicated Google map service channeled from the domain http://www.google.cn/maps, which could display the map titles correctly in China. Is it possible to modify the code a bit to accommodate this map service for China?

I'm not experienced with programming actually, so have to ask your time to help. Thanks!

CrazyDude1994, 2019-05-12:
Hi. I've looked what can be done, but you probably know that China uses some other coordinate system. Because of it, it has some offset. I'm not sure what I can do now in order to fix it. I saw there are algorithm which may decode coordinates, but it will take some time to me to check them.

rivermtnn, 2019-05-15:
Hi, thanks for getting back on this.

I know there is another coordinate system in China. I will try to dig more information over this topic and bring back.

rivermtnn, 2019-05-15:
Hi thanks for the time to look into this case.

I've run through several pieces of info on the implication of maps in China, I believe you have already read them also.

The causes of this trouble are two as my understand:

1. The access to the standard Google map service is blocked in China by the national firewall.
2. Map data served by any publicly accessible API in China is using the projection datum called "GCJ-02", which is a none-linear deviation from global standard WGS-84.

So loading the app in China, first you will see a blank map view, nothing except the marker of the phone and drone. If try run away from the firewall, then you could see the map tiles in the view port. However the makers are displayed at the wrong locations on the map view. Because of that makers are set with the actual WGS84 GPS coordinates from the telemetry data stream, and the map data loaded from Google map service is encoded with "GCJ-02".

Just my imagination, to make the app function in China as intended, there are two steps needed to address above issues:

1. First, find a way to load the map data from the service located https://google.cn/maps/. However it seems there is now way to direct the Google Map SDK to use any similar map data source a like. 

The only relevant official info I could find is in the Wear OS SDK: https://developer.android.com/training/wearables/apps/creating-app-china. It mentioned two points: first to use FusedLocationProvider API to get the current device location coordinates. And the coordinate reported by FusedLocationProvider API is based on WGS-84, so conversion to GCG-02 is needed to display with any other SDKs.

So basically it means there is Maps SDK support in China. I have seen any app with such function too. One way I thought is to use the JS API from the https://google.cn/maps/ to display the map in the app, but there could be too much change to the current code. The other way is to use other map services like mapbox, OSM or Baidu, it also will take much time. I do find a lib to implement such function, maybe a time saver: https://github.com/sharenowTech/AnyMaps.

2. Second, convert the WGS-84 coordinate from telemetry to GCJ-02 just for display. As you said, the lib is available here: https://github.com/googollee/eviltransform. 

Also Baidu API is accessible like: http://api.map.baidu.com/ag/coord/convert?x=121.583140&y=31.341174&from=0&to=2&mode=1. The output is base64 encoded.

Since it's impossible to use Google Maps SDK to get map data to the app without shorting the firewall, it seems the better direction is to implement other map service. And time costing. If using other map service, I think there is no need to convert coordinates outside the SDK.

Yeah, it's easier just add a user setting option to enable the coordinate conversion, without changing other parts. And the app will function as intended in China, just need to avoid the firewall at the same time.

CrazyDude1994, 2020-02-19:
Hey @rivermtnn 
Sorry for that long delay. Do you still use my application? I would like a little assistance from your side