CrazyDude1994/android-taranis-smartport-telemetry #34: USB option

KFNCOLO, 2019-06-06:
I should have realized this earlier, but I'm using 2.4 ghz for my fpv video and while the bluetooth connection doesn't impact the video alot, it does impact it. I am really bummed since I have been loving this real time tracking (and after the fact crashed aircraft finding) capability.

Is there any possibility of taking telemetry feed from the inverter and instead of going into a bluetooth module the inverted signal would instead be wired to a USB cable that could be connected to the phone, eliminating the bluetooth connection? I am totally ignorant on whether this is even possible, but I'm guessing this would need to be a software/programming change where there would be an option of selecting either bluetooth or USB.  Anyway, just an idea.

And again, many thanks to you for this incredible app.

CrazyDude1994, 2019-06-07:
I think it's possible. But as you said of course it will require code modification. Also it's probably going to be Transmiter -> Inverter -> USB TTL Serial -> [probably OTG] -> phone

CrazyDude1994, 2019-11-09:
@KFNCOLO hi.

It took a while for this feature to arrive, but now I'm happy to announce that I made a beta version. I need your feedback about it. More info is at https://github.com/CrazyDude1994/android-taranis-smartport-telemetry/issues/47

I'm closing this issue, please write your opinion on the linked issue. Thanks!

closed 2019-11-09 by CrazyDude1994