CrazyDude1994/android-taranis-smartport-telemetry #36: Built in bluetooth on QX7S repeatedly disconnects.

maxduracell, 2019-07-26:
I downloaded your release candidate from the built in Bluetooth thread.  I could not make a connection until I powered up a bound FC.  Once I did, the connection was quick and I was able to verify connection by changing flight modes, but it only lasted around 30 seconds before I got an error that the PIN was not accepted.  I'm using a QX7S on OpenTX 2.2.3, R9M, R9MM on F.port.  I'm pretty new the this, but I would love to help.

CrazyDude1994, 2019-08-03:
You need to disable pincode. 

AT+TYPE0 is the command

CrazyDude1994, 2019-08-03:
Nvm, I read you have built in bluetooth. I remember one guy had same issue, but I believe he fixed it somehow

CrazyDude1994, 2019-08-03:
Here is his message: "Quick disconnection from BLE is almost certainly related to Android's behavior - on my OP5T with Oxygen OS (Pie) you have to accept pairing twice as I reported on GH, but on other ROMs it might be different"

So I guess you should try to play with pincode entering, You can pair it first for example manually in bluetooth settings

CrazyDude1994, 2019-08-03:
Also found this: "After I did some research and testing, I found that the automatically disconnection problem has something to do with the interference between Bluetooth and WIFI on Nexus 7. If I turned off the WIFI, then the disconnection problem of Bluetooth has gone. And this problem did not occur on Galaxy 3,4,5." You can try this too

CrazyDude1994, 2019-09-10:
Closing because there is no more interest

closed 2019-09-10 by CrazyDude1994

brainbubblersbest, 2020-03-27:
Had same issue with Horus x12s   .f. Port r9m
Can only connect with established connection to FC. 
Than connection to Horus will last only 30seconds.

Btw. In opentx settings only baud i can decide is 115200 and 400.000????
I cannot change the pin also? Only the bt name