CrazyDude1994/android-taranis-smartport-telemetry #42: Telemetry over TCP/IP or UDP

ramoonas, 2019-09-30:
I guess a feature request if possible and willing. As the name implies would be helpful to have an option to connect to specfici IP over TCP or UDP via Wifi.

danielolsman, 2020-05-22:
you mean a connection to a esp-01? like esp link? https://github.com/jeelabs/esp-link

ramoonas, 2020-05-22:
I am not exactly sure how esp link works but i have a transparent bridge made which listens for tcp connections on specific port and just forwards whatever it receives via smartport to tcp client. So it would need to just open a connection to specific ip on specific port and that's it. Not sure if that is something you want to work on

techfixpros, 2020-12-24:
If this is within the scope of OP's request, there is an android app called MavLinkHub that receives Mavlink to my phone with a telemetry radio connected, then pushes the Mavlink telemetry to my laptop running MissionPlanner over WiFi using UDP (supports TCP as well). I too would like to see this functionality for S.Port telemetry, but don't know of a ground station software that would be able to use it...

KlemenDEV, 2021-02-05:
Crossfire also transmits mavlink on UDP port on its WiFi, this feature would come in handy there.