CrazyDude1994/android-taranis-smartport-telemetry #45: stopped working with Taranis x9d+ 2019, R9M-2019 access protocol

Mmarsian, 2019-11-05:
Hi, I changed my Taranis and R9M to the 2019 versions with access. Though I have telemetry on my Taranis and the Bluetooth is connected there are no data anymore in the android app.
It seems as there is no communication on the s.port of the R9M2019.
Somewhere I was reading that there is no incresed speed for over the air communication. Can it be that they have changed baud rate on the s.port of the R9M2019?

CrazyDude1994, 2019-11-05:
Well, I don't know. I don't have new taranis. Why you just don't try to reconfigure your bluetooth module to the higher baud rate? It's probably 115200 now

Mmarsian, 2019-11-06:
I'll try that. I suppose they want to combat Crossfire. I was reading something like 450k. Hopefully the Bluetooth module can handle it.

Mmarsian, 2019-11-08:
Well, I tried every Baudrate from 57600 to 460800 but without success. Bluetooth connection is ok, there is something coming out of the s.port (verified with ftdi adapter to my pc, but I had no inverter) but nonetheless no values in the app.
I hear that Access is encrypted, can it be that the s.port on the R9M-2019 has encrypted data?

CrazyDude1994, 2019-11-09:
> Well, I tried every Baudrate from 57600 to 460800 but without success. Bluetooth connection is ok, there is something coming out of the s.port (verified with ftdi adapter to my pc, but I had no [...]

Best way to tell is you to connect your ftdi to smart port with an inverter and record some traffic and send it here so I can take a look at data to understand it

Mmarsian, 2019-11-12:
Hello, I opened an issue at opentx:
https://github.com/opentx/opentx/issues/7076#issue-521212833
They confirmed the Baudrate of 450k which I think is exactly 460800, at least the HC06 can be only configured with AT+BAUDA.
Now there is not only telemetry coming in the downlink, but a whole lot more. I dont't know if the telemetry frame can be filtered out, Transmitting the whole 450k datasream to the smartphone app and then doing the filtering seems a bit gross for me, but anyhow that's far beyond my abilities.

CrazyDude1994, 2019-11-16:
well you can try to record a log of data which now comes from ACCESS and I will try to look at it and understand what I can do about it

skaman82, 2020-05-18:
@CrazyDude1994 I just did a capture with logic from my HM10 connected to my X-LITE BT module. I don't know if that helps
[Logic Capture.zip](https://github.com/CrazyDude1994/android-taranis-smartport-telemetry/files/4644929/Logic.Capture.zip)

Mmarsian, 2020-05-20:
@skaman82 in my last post I mentioned my issue I opened with opentx bc I wanted to find out the baudrate on the R9-2019 external module. I think they confirmed that it's 450k. It must be increased from the original 57600 because the telemetry downstream is much faster now. With that being said I would doubt that capturing at 57600 baud is helpful, but I'm not sure.

skaman82, 2020-05-20:
@Mmarsian here the capture @ 115200. My HM10 can't do 450k. I set the baud-rate to 115200 (default) on my x-lite also.
[115200_log.zip](https://github.com/CrazyDude1994/android-taranis-smartport-telemetry/files/4656648/115200_log.zip)

Mmarsian, 2020-05-20:
@skaman82 for me it seems strange that it works with Crossfire, as Crossfire is also using a much higher rate on the airlink. Which baudrate you use with crossfire?

skaman82, 2020-05-25:
@Mmarsian crossfire is working at 115200 for me

StefanKellerAC, 2020-05-29:
find my solution with an X10 and ACCESS upgrade module here:
https://www.rcgroups.com/forums/showpost.php?p=44636915&postcount=1005

skaman82, 2020-06-05:
@StefanKellerAC I just wrote you a PM on rcgroups :D

Mmarsian, 2020-11-16:
> find my solution with an X10 and ACCESS upgrade module here:
> https://www.rcgroups.com/forums/showpost.php?p=44636915&postcount=1005

@StefanKellerAC Does this solution help in the way to produce a telemetry stream which can be serve as an input to @CrazyDude1994 App?

StefanKellerAC, 2020-11-16:
yes, with a teensy you could deliver it as you want :-)