CrazyDude1994/android-taranis-smartport-telemetry #48: How to open/view log files?

rsilk1949, 2020-02-02:
How can I open/read the .LOG files?
Every .CSV file is always 0 bytes??

Enhancement :
Pls add a setting to allow the user to specify the path of the directory for the log files.

rsilk1949, 2020-02-02:
Extra info: all tests done on the bench with good gps fix.

CrazyDude1994, 2020-02-02:
You can review .log files by pressing replay button which is at the Settings button when you are not connected to anything. CSV files is just an old feature that I forgot to remove

rsilk1949, 2020-02-03:
Is it possible to view the actual contents of the log files with any particular Android app? I am trying to debug why the mAh is not displaying correctly on my radio. It is correct in your app!

rsilk1949, 2020-02-03:
I have found "WindHex" app to view the log files and try to debug why the mAh is correctly displayed in your app but not on my radio's telemetry page.
Does your log file contain raw smartport telemetry data?

CrazyDude1994, 2020-02-03:
Yes, it's raw smart port data

rsilk1949, 2020-02-12:
The mAh values from iNav are sent in the Fuel data packets. Displaying the Fuel values on the radio produced the expected mAh numbers.

closed 2020-02-12 by rsilk1949