CrazyDude1994/android-taranis-smartport-telemetry #51: Feature request: add playback controls

rsilk1949, 2020-02-14:
Currently there is only a  control which can be dragged horizontally to replay a log file.
Suggest adding plackback controls (e.g. play, pause, stop, etc.)

rsilk1949, 2020-02-15:
Actually, it only needs a single pause/play control  below the slider control (c.f. iPlayer).

DiscoMan18, 2020-05-13:
Also if you gonna keep that playback slider, please change its color and size, as currently it's almost unusable. Another issue that migrated from 1.5 to 1.7 is the issue when during playback on the map appear random black lines.

Also would appreciate if the map will be automatically centered and zoomed in to initial aircraft location at the start of playback

druckgott, 2021-06-26:
Is there already a play button I can't see a Play button. I can only move manually with my finger. A Play paus stop Bütten will be quite nice the app is really cool