CrazyDude1994/android-taranis-smartport-telemetry #53: Jumper t16 with crsf help

darrenwest7, 2020-03-03:
Hi I've previously used the app on my horus x12s and r9 system with perfect results. I've since upgraded to a t16 and full crossfire module. I still have my bt module i made for the r9 is there any way I can get this to work with my current setup to view the crsf telemetry maybe solder the bt module and inverter direct to the pins inside the jumper? 
Thanks all

CrazyDude1994, 2020-03-07:
@darrenwest7 hi. Unfortunately I don't own Jumper T16. But it seems that you can modify it to use bluetooth module. You can google about it. There is also some link https://www.rcgroups.com/forums/showpost.php?p=43107877&postcount=163

darrenwest7, 2020-03-07:
Hi thanks for the reply from what I understand the t16 works just like any other frsky radio and ideally I want to use the higher power HC-05 module I have rebuilt and ready to go. I'm just a little stuck as to what information and which format and baud rate it will be coming from the main board solder points for the x12 bluetooth module. Will I still need an inverter on there if it's been changed by the radio to s.port format or does it just get passed through directly from the crsf module to the tx/rx pins. 

Hopefully you or someone else may have some of the answers 

Thanks darren

darrenwest7, 2020-03-09:
Ok I give up I'm just going to order the horus x10 module from the uk supplier and use that

closed 2020-03-12 by darrenwest7

dentex, 2021-03-17:
> Ok I give up I'm just going to order the horus x10 module from the uk supplier and use that

Hello. How did it go? Because it seems it cannot be connected at all.