CrazyDude1994/android-taranis-smartport-telemetry #54: Connects over BT but no data received

maciek252, 2020-03-21:
I've done everything by the book, I'm running a Taranis X9D+ with a relatively recent OpenTX and two FrSky links: either the internal one or a R9M. I've soldered the BT inverter. The telemetry in Taranis works (yaapu LUA script). However unfortunately I get no telemetry on an Android phone. I've connected the BT dongle either to the internal Taranis serial ("telemetry mirror" setting enabled) or to the R9M's S.port.

I've tried both the program directly from Google Play site (stable and beta) and a version compiled by myself. Here, in the debugger, I get the following messages:

03-21 17:06:10.195 3435-3598/crazydude.com.telemetry D/FrSky Protocol: Unknown packet[27, 16, 6, 80, 58, 43, 14, 0, 38]
03-21 17:06:10.226 3435-3598/crazydude.com.telemetry D/FrSky Protocol: Unknown packet[27, 16, 3, 80, 6, 32, 34, 0, 84]
03-21 17:06:10.266 3435-3598/crazydude.com.telemetry D/FrSky Protocol: Unknown packet[27, 16, 5, 80, 0, 0, 66, 5, 83]
03-21 17:06:10.184 3435-3598/crazydude.com.telemetry D/FrSky Protocol: GPS: 0
03-21 17:06:10.430 3435-3598/crazydude.com.telemetry D/FrSky Protocol: Unknown packet[27, 16, 4, 80, 0, 0, 0, 0, 155]
03-21 17:06:10.437 3435-3598/crazydude.com.telemetry D/FrSky Protocol: Unknown packet[152, 16, 4, 241, 88, 0, 0, 0, 161]

What's wrong here?

CrazyDude1994, 2020-03-22:
We don't support Ardupilot passthrough telemetry. Only supported telemetry is from INAV/Betaflight

maciek252, 2020-03-22:
Thanks for clarification, I think that it could be stated clearly in the documentation which FrSky telemetry standard is supported and which isn't (as what you refer to as "Ardupilot passthrough telemetry" is in fact also FrSky's?)

I've found a library for parsing FrSky passthrough messages and will try to add it to the app:

https://github.com/zs6buj/Frsky2Mavlink

maciek252, 2020-03-27:
So I've got a running version. Message ID's and int values are decoded correctly (for example, num of sats or flight mode), float values sometimes have seem to have a problem, but I don't know yet. Still examining. Are you interested in a pull request or sending the files?

CrazyDude1994, 2020-03-30:
@maciek252 sure, pull request would be great. But it would be better if you could float values problem. But you still can make a pull request and maybe I will have an idea.

maciek252, 2020-04-06:
I think there is a bug in the implementation of bit32Extract already in the C++ Arduino sketch. I've described it here:

https://github.com/zs6buj/Frsky2Mavlink/issues/2

Interesting that lua provides this implementation but neither C++ nor Kotlin does.