CrazyDude1994/android-taranis-smartport-telemetry #56: R9M2019 issue

zosko, 2020-05-03:
Did have some success to connect with R9M2019 with last ACCST firmware ? maybe is different baud rate ?

CrazyDude1994, 2020-05-04:
I don't believe that someone successfully connected it with ACCST. It seems like it's completely different protocol which is not supported for now

zosko, 2020-05-04:
@CrazyDude1994 R9M have ACCESS and ACCST firmware, both support S.Port, but in both i did not have any success.

What firmware to use for R9M so i can use S.Port 3 pins on him ?

hyperion11, 2020-05-16:
I have R9M2019 fw from 2019-02-13 version 190117.
R9MM receiver also ACCST with latest fw. It works with betaflight

skaman82, 2020-05-18:
same issue: xlite with buld-in BT module transmitting telemetry to a HM10 module.
Crossfire telemetry is working with no issues. R9M Lite Pro (ACCESS) is not working. :(

closed 2021-04-15 by zosko