CrazyDude1994/android-taranis-smartport-telemetry #60: Ardupilot \ Arducopter \ Arduplane support?

hyperion11, 2020-05-16:
hello! Thanks for your app! It great!
I make BT adapter for R9M + R9MM (latest FCC non ACCESS FW) and check it with betaflight 4.1.6. 
It work like a charm (I connect inverted S.Port(double inverted) pin on R9MM to Uart3 Omnibus Pro V3).
Than try connect another R9M + R9MM (same FW) to Pixracer with Arduplane 4.0.5. I use didicated FRSKY port on pixracer and connect as shown on this picture
[image]
i found schematic for pixracer i think here inverter
[image]
so i think inverted S.port signal is ok
next i configure post as discribed here
https://ardupilot.org/copter/docs/common-frsky-telemetry.html
here my setting for this this port
[image]
after all with Ardupilot Telemetry Viewer not working. But should it work?

hyperion11, 2020-05-16:
https://1drv.ms/u/s!AuFPVI_eDaZbhMZxGDKkwAHZdUYbWw?e=qexVYm
here log collected by bt_debug-debug.apk

CrazyDude1994, 2020-05-16:
No, Ardupilot/arducopter/ whatever is not supported. Currently fully supported is inav. Because it's very similiar to BF, bf works too.

hyperion11, 2020-05-17:
bad :(
I try to understand how it works. I grab two logs, from BF and Ardu. Here BF log 
7E67
7E1B100650749B0A007F
7E48
7E1B1002503C068018C2
7EE9
7E6A
7ECB
7E1B100650749B0A007F
7E1B100750E40C0005A2
7EAC
7E0D
7E1B100650749B0A007F
7E8E
7E1B100650749B0A007F
7E981004F156000000A3
7E981001F16400000098
7E1B100008040B458310
7E1B1000088ADCEB0193
7E2F
7E1B100650749B0A007F
7ED0
7E1B1005500004420153
7E71
7EF2
7E1B100650749B0A007F
7E53
7E34
7E1B1003500004020096
7E1B1004500440005502
7E1B100650749B0A007F
7E95
7E16
7EB7
7E1B100650749B0A007F
7E39
7EBA
7E00
7E1B100650749B0A007F
AFAIK, 7E - start byte. next byte - ID sensor, than always "10"  byte, than two bytes - ID Data, than 4 byte DATA, and CRC byte at end, is correct?
i look into this source
https://github.com/CrazyDude1994/android-taranis-smartport-telemetry/blob/master/app/src/main/java/crazydude/com/telemetry/protocol/FrskySportProtocol.kt
and 
https://github.com/opentx/opentx/blob/2.3/radio/src/telemetry/frsky.h
and cannot find any ID's from this log. Where am i wrong?

hyperion11, 2020-05-26:
PR added

closed 2020-06-02 by hyperion11