CrazyDude1994/android-taranis-smartport-telemetry #61: Unstable connection

skaman82, 2020-05-18:
I get the app to connect to my x-lite but the connection is breaking up after about 10-20 Seconds every time. On the other hand if I connect a HM10 to my x-lite - the link is stable. Any ideas why the App is disconnecting every time? 

• xlite (horus12 bt module added in)
• Android Phone (Huawei 20P lite)

kniuk, 2020-05-19:
My connection gets broken unless I click "pair" on my phone's pop up and
then click pair again and enter pincode (000000). it happens every time I
initiate connection. after such pairing connection is stable. jumper t16
with frsky bt module from horusrc, mi 9tpto phone, r9m 2019 with flex accst
fw.

CrazyDude1994, 2020-06-09:
@skaman82 can you please try @kniuk comment?

CrazyDude1994, 2020-06-11:
There is no such thing as 'baud rate' when you connect using bluetooth. Baud rate used when connecting by wire

closed 2020-06-11 by CrazyDude1994