CrazyDude1994/android-taranis-smartport-telemetry #70: Crossfire 3.72 battery sensor not read

skaman82, 2020-06-24:
I have some empty data in the app using crossfire fw 3.72. 

Missing: 
• Bat voltage
• Consumed mAh
• Amp draw

Build-In BT module of the X-lite set to telemetry. INAV on the Plane.

CrazyDude1994, 2020-06-26:
So it was working pre 3.72, and stopped working after 3.72?

skaman82, 2020-06-28:
@CrazyDude1994 yes, with the FW 2.42 it was working fine

avs2, 2020-10-15:
I just ran into this problem as well. I am using Crossfire 3.78 and a TX16S. Telemetry is correctly received by the radio but battery stats are not displayed in the app. I assume TBS changed something in how these values are named in the telemetry stream. Here is a log file: [btlog.zip](https://github.com/CrazyDude1994/android-taranis-smartport-telemetry/files/5387605/btlog.zip).

[image]

[image]

HazzaHFPV, 2020-10-30:
I tested all Crossfire versions from 2.93 up to 4.0.6 and can confirm the problem. Sensors are not showing, only GPS data seems valid.
Crossfire Version 2.42 funnily enough would make it impossible to pair with the bluetooth device of my T16, which I don't quite understand.

avsaase, 2020-12-20:
I'm also not getting distance to home. I've read there is a bug in OpenTX 2.3.10 that causes problems with Crossfire telemetry sensors, not sure if that has anything to do with this.

CrazyDude1994, 2021-05-13:
Fixed in https://github.com/CrazyDude1994/android-taranis-smartport-telemetry/issues/113

closed 2021-05-13 by CrazyDude1994