CrazyDude1994/android-taranis-smartport-telemetry #71: RSSI and custom audio warnings

HazzaHFPV, 2020-07-12:
Not an issue, more a begging for features. I'd love to be able to display my FrSky RSSI in some way. I always imagined it being a simple sensor value of some kind which is supplied by the receiver. Could this be added somehow?
Secondly, would it be possible to add custom warnings by letting the user select a sensor and define a threshold so that a warning is played when the value drops below the threshold? Would be useful for RSSI, voltage and altitude for example.
This is just an idea. It is probably much harder to add than I can imagine, so if it seems cheeky I apologize.

Apart from that I do have to thank you for this awesome app. It has bought new life to my old PPM transmitter using a 4in1 multiprotocol module with an HC-06 bluetooth module attached (no further modding required). The radio does not have the capability to display any telemetry, so being able to send the data to my android phone is a huge plus.

I know that requests for new features from people like me is probably a hassle, but I do not have the knowledge to add them myself or to contribute in any other way. (Maybe you should add a paypal donation link to this github project as well :))

CrazyDude1994, 2020-07-19:
It already has been requested a few times. I'm working on the audio warning. I'm not sure about RSSI tho, because last time I checked TX didn't send RSSI as a telemetry data (atleast for the R9M)

HazzaHFPV, 2021-05-17:
I know it is cheeky, but i would like to bring this topic up one more time. Unfortunately I have still not learned how to code or even compile android apps to try and do this myself, but as input to this project has increased I thought I might raise the question one last time regarding the display or RSSI.
The RSSI sensor seems to be defined within the FrSky SPORT telemetry and it's address seems to be present within the FrskySportProtocol.kt as "const val RSSI_SENSOR = 0xF101".
As I am incapable of doing it myself, would it not be possible to display the value of this sensor the same way the other sensors are displayed? I am really sorry that I am unable to contribute anything meanungful to this topic.
@RomanLut I have seen you contributing great work, maybe you have an idea?
Thank you very much everyone for the hard work and this great app

On a different note, maybe someone can give me a hint: How and where would I be able to find information and learn to contribute to such projects? Where would I start? Thanks again, everyone

RomanLut, 2021-05-24:
Please help with testing.

RSSI display is implemented in dev build:
https://github.com/RomanLut/android-taranis-smartport-telemetry/blob/master/app-debug.apk

I can not test it myself with SPORT because I am using mavlink.
If it works, I will submit pool request to this repository.

HazzaHFPV, 2021-05-25:
Man you are awesome. Thank you for putting the work in. Unfortunately it is only displaying a dash for RSSI and no value.
All other sensors are working fine. Also the google maps are not loading, OpenStreetMap is, however, working fine.

I have tested with an bluetooth module connected to my old radio which is currently running a mulit-protocol module in PPM mode and I have tested it with a Jumper T16 with internal Multi-Module and the Horus X10 bluetooth chip installed with the same result:
All sensors are working, but RSSI displays horizontal dash. I have attached a picture of the telemetry screen of my Jumper, just in case the sensor ID might help.

I am very grateful for all the effort you guys are putting into this and sorry that at this time I am unable to contribute anything useful myself except a small donation over patreon.
[image]
[image]

RomanLut, 2021-05-25:
Please share a log file. It contains raw telemetry stream. This way I might find problem with Frsky protocol decoder.

HazzaHFPV, 2021-05-25:
Ah yes. Sorry for not providing it in my last post. Have a look at attached file, I hope it helps
[Telem Log.zip](https://github.com/CrazyDude1994/android-taranis-smartport-telemetry/files/6541684/Telem.Log.zip)

RomanLut, 2021-05-26:
Please confirm RSSI works for you in this build:
https://github.com/RomanLut/android-taranis-smartport-telemetry/blob/master/app-debug.apk

p.s. Google Maps and RC channels widget should not work in this build, please use OpenStreetMaps.

HazzaHFPV, 2021-05-27:
Inital testing in and around the house shows that RSSI seems to work with this build :)
I will do some more testing while flying on the weekend. Thank you ever so much for the work and for including this.

On a side-note, but maybe not relevant: this build seems to crash occasionally when the screen is rotated.
However, I am really thrilled ^^ This app is really breathing new life into my old PPM radio :)

RomanLut, 2021-06-02:
Submitted pull requests for rrsi sensor:
https://github.com/CrazyDude1994/android-taranis-smartport-telemetry/pull/123
https://github.com/CrazyDude1994/android-taranis-smartport-telemetry/pull/124
https://github.com/CrazyDude1994/android-taranis-smartport-telemetry/pull/125
https://github.com/CrazyDude1994/android-taranis-smartport-telemetry/pull/126

HazzaHFPV, 2021-06-02:
I finally got around doing my last tests while really flying today.
Worked like a charm :) thank you very much