CrazyDude1994/android-taranis-smartport-telemetry #76: Feature request: CSV log files

rsilk1949, 2020-09-30:
I believe that at some time log files were saved in both LOG and CSV file types.
Pls enable the saving of log files in CSV so we can easily analyze the flight data afterwards.
e.g. on the Settings page :

Telemetry files:
Logging: (Y/N)
Path: <filepath>
CSV: (Y/N)

marianomd, 2023-04-14:
Hello @RomanLut I'm implementing this in a fork of your fork. I'm having a hard time trying to compile it. 

Can you tell me which NDK version you are using?

```
> Task :libuvccamera:ndkBuild FAILED
fcntl()make: Entering directory 'C:/Users/USER/Documents/git/android-taranis-smartport-telemetry/libuvccamera/src/main'
Android NDK: android-14 is unsupported. Using minimum supported version android-16.    
Android NDK: WARNING: APP_PLATFORM android-16 is higher than android:minSdkVersion 1 in ./AndroidManifest.xml. NDK binaries will *not* be compatible with devices older than android-16. See https://android.googlesource.com/platform/ndk/+/master/docs/user/common_problems.md for more information.    
: Bad file descriptor
make: Leaving directory 'C:/Users/USER/Documents/git/android-taranis-smartport-telemetry/libuvccamera/src/main'
make: Entering directory 'C:/Users/USER/Documents/git/android-taranis-smartport-telemetry/libuvccamera/src/main'
fcntl()[x86] Install        : libusb100.so => libs/x86/libusb100.so
: Bad file descriptor
make: Leaving directory 'C:/Users/USER/Documents/git/android-taranis-smartport-telemetry/libuvccamera/src/main'
fcntl()make: Entering directory 'C:/Users/USER/Documents/git/android-taranis-smartport-telemetry/libuvccamera/src/main'
: Bad file descriptor
[x86] SharedLibrary  : libjpeg-turbo1500.so
ld: error: can't create dynamic relocation R_386_32 against local symbol in readonly segment; recompile object files with -fPIC or pass '-Wl,-z,notext' to allow text relocations in the output
make: Leaving directory 'C:/Users/USER/Documents/git/android-taranis-smartport-telemetry/libuvccamera/src/main'
>>> defined in ./obj/local/x86/libjpeg-turbo1500_static.a(jfdctflt-3dn.o)
>>> referenced by jni/libjpeg-turbo-1.5.0/simd/jfdctflt-3dn.asm
>>>               jfdctflt-3dn.o:(.text+0xB4) in archive ./obj/local/x86/libjpeg-turbo1500_static.a

ld: error: can't create dynamic relocation R_386_32 against local symbol in readonly segment; recompile object files with -fPIC or pass '-Wl,-z,notext' to allow text relocations in the output
>>> defined in ./obj/local/x86/libjpeg-turbo1500_static.a(jfdctflt-3dn.o)
>>> referenced by jni/libjpeg-turbo-1.5.0/simd/jfdctflt-3dn.asm
>>>               jfdctflt-3dn.o:(.text+0xF5) in archive ./obj/local/x86/libjpeg-turbo1500_static.a

ld: error: can't create dynamic relocation R_386_32 against local symbol in readonly segment; recompile object files with -fPIC or pass '-Wl,-z,notext' to allow text relocations in the output
>>> defined in ./obj/local/x86/libjpeg-turbo1500_static.a(jfdctflt-3dn.o)
>>> referenced by jni/libjpeg-turbo-1.5.0/simd/jfdctflt-3dn.asm
>>>               jfdctflt-3dn.o:(.text+0x104) in archive ./obj/local/x86/libjpeg-turbo1500_static.a

ld: error: can't create dynamic relocation R_386_32 against local symbol in readonly segment; recompile object files with -fPIC or pass '-Wl,-z,notext' to allow text relocations in the output
>>> defined in ./obj/local/x86/libjpeg-turbo1500_static.a(jfdctflt-3dn.o)
>>> referenced by jni/libjpeg-turbo-1.5.0/simd/jfdctflt-3dn.asm
>>>               jfdctflt-3dn.o:(.text+0x10C) in archive ./obj/local/x86/libjpeg-turbo1500_static.a

ld: error: can't create dynamic relocation R_386_32 against local symbol in readonly segment; recompile object files with -fPIC or pass '-Wl,-z,notext' to allow text relocations in the output
>>> defined in ./obj/local/x86/libjpeg-turbo1500_static.a(jfdctflt-3dn.o)
>>> referenced by jni/libjpeg-turbo-1.5.0/simd/jfdctflt-3dn.asm
>>>               jfdctflt-3dn.o:(.text+0x114) in archive ./obj/local/x86/libjpeg-turbo1500_static.a

ld: error: can't create dynamic relocation R_386_32 against local symbol in readonly segment; recompile object files with -fPIC or pass '-Wl,-z,notext' to allow text relocations in the output
>>> defined in ./obj/local/x86/libjpeg-turbo1500_static.a(jfdctflt-3dn.o)
>>> referenced by jni/libjpeg-turbo-1.5.0/simd/jfdctflt-3dn.asm
>>>               jfdctflt-3dn.o:(.text+0x204) in archive ./obj/local/x86/libjpeg-turbo1500_static.a

ld: error: can't create dynamic relocation R_386_32 against local symbol in readonly segment; recompile object files with -fPIC or pass '-Wl,-z,notext' to allow text relocations in the output
>>> defined in ./obj/local/x86/libjpeg-turbo1500_static.a(jfdctflt-3dn.o)
>>> referenced by jni/libjpeg-turbo-1.5.0/simd/jfdctflt-3dn.asm
>>>               jfdctflt-3dn.o:(.text+0x24B) in archive ./obj/local/x86/libjpeg-turbo1500_static.a

ld: error: can't create dynamic relocation R_386_32 against local symbol in readonly segment; recompile object files with -fPIC or pass '-Wl,-z,notext' to allow text relocations in the output
>>> defined in ./obj/local/x86/libjpeg-turbo1500_static.a(jfdctflt-3dn.o)
>>> referenced by jni/libjpeg-turbo-1.5.0/simd/jfdctflt-3dn.asm
>>>               jfdctflt-3dn.o:(.text+0x25A) in archive ./obj/local/x86/libjpeg-turbo1500_static.a

ld: error: can't create dynamic relocation R_386_32 against local symbol in readonly segment; recompile object files with -fPIC or pass '-Wl,-z,notext' to allow text relocations in the output
>>> defined in ./obj/local/x86/libjpeg-turbo1500_static.a(jfdctflt-3dn.o)
>>> referenced by jni/libjpeg-turbo-1.5.0/simd/jfdctflt-3dn.asm
>>>               jfdctflt-3dn.o:(.text+0x262) in archive ./obj/local/x86/libjpeg-turbo1500_static.a

[... cut ...]
[arm64-v8a] Install        : libusb100.so => libs/arm64-v8a/libusb100.so
make: Leaving directory 'C:/Users/USER/Documents/git/android-taranis-smartport-telemetry/libuvccamera/src/main'
fcntl()make: Entering directory 'C:/Users/USER/Documents/git/android-taranis-smartport-telemetry/libuvccamera/src/main'
: Bad file descriptor
[arm64-v8a] Install        : libuvc.so => libs/arm64-v8a/libuvc.so
make: Leaving directory 'C:/Users/USER/Documents/git/android-taranis-smartport-telemetry/libuvccamera/src/main'

```

marianomd, 2023-04-15:
I implemented the functionality and created a PR on @RomanLut 's fork. 

https://github.com/RomanLut/android-taranis-smartport-telemetry/pull/40