CrazyDude1994/android-taranis-smartport-telemetry #77: X9D 2019 non-Se: no telemetry in app but bluetooth connects

avs2, 2020-10-04:
Hi, thanks for making this app. I really hope to get with working with a X9D+ 2019 (non-SE) and Crossfire Nano Tx, connecting the bluetooth module to the s-port in the battery bay. I am able to connect the app via bluetooth, but the only telemetry that is shown is zero satellites. Since I am indoors I don't know if that is actual data or just a placeholder value. The logs created by the app are all 0 byte files.

What I have done/noticed so far:
- Before soldering I set the baud rate to 57600.
- I think I have wired everything up correctly:
[image]
[image]
As you maybe noticed this bluetooth module looks slightly different from the one on the home page. A friend had this one laying around so I don't know the exact type. Could that be the problem? 
- There is no voltage on the VCC pin in the battery bay. For this test I powered the module from a separate 5V supply, so I think that is not the problem though.
- I don't have a "mirror s-port" type option in the hardware menu of my Taranis (OpenTX 2.3.3). I do have a Bluetooth option, but I think this transmitter doesn't have an internal bluetooth module, so I left that disabled.

Maybe you can spot the problem?

CrazyDude1994, 2020-10-05:
If logs are 0 size, it means it doesn't output any data.

avs2, 2020-10-06:
I just learned in https://github.com/opentx/opentx/issues/5357 that it is not possible to mirror crossfire telemetry to the Smartport in the battery bay of the 2019 X9D, only the UART in the old X9D. Bummer.

closed 2020-10-06 by avs2