CrazyDude1994/android-taranis-smartport-telemetry #78: [FR] Can the Multiprotocol module telemetry be added?

Tygrys-1, 2020-10-10:
There are many new radios like Radiomaster TX16S and Jumper T16, T18, with multiprotocol module. They do not have classic SPort output. Ok, they have it in the module bay, but it is not convinient to use from there. Multiprotocol module uses it's own telemetry format, based I think on FRSky, but kinda simpler. It receives SPort telemetry from FRSky receivers, but outputs it to the radio in it's own format.

Radiomaster TX16S has two serial ports on the bottom and HC06 can be connected to any of them. The "Telemetry mirror" works on 115200 Baud and the signal is not inverted. I can capture the BT signal on the PC to the file, and it is received the same way on the phone by the Telemetry Viewer application.
It is the same iNav information, just coded a little bit different, sensors have 16 bit id's and usual 6 bytes of sensor data.

Is it possible to decode this in the Telemetry Viewer application? 
Sample iNav telemetry log file here:
[2020-10-09 22-03-16.zip](https://github.com/CrazyDude1994/android-taranis-smartport-telemetry/files/5359299/2020-10-09.22-03-16.zip)

stefaanbolle, 2021-01-18:
How did you connect and enable bluetooth on the TX16S?

Tygrys-1, 2021-01-18:
On System-hardware page, set the Serial Port (TTL) to "Telem mirror"
For hardware connection, connect the bottom port to the Bluetooth like HC 05 or HC 06, GND->GND, +5V->+5V, TX->RX and RX->TX.
Earlier you have to configure the bluetooth module through your computer for 115200 Baud, no parity, 8 bits and one stop bit. Use serial port or USB to serial dongle and any terminal emulating program.

stefaanbolle, 2021-01-18:
Thanks!
So no S.Bus inverter needed.
Can it be connected to the external module bay uart?

Tygrys-1, 2021-01-18:
You can use module bay, but there you will need the inverter. 
BTW - you do not need the radio RX-TX bluetooth connection on bottom port, as the communication is only one way.

stefaanbolle, 2021-01-18:
Would you mind to post a schematic how you connect it to the internal module bay or external uart?
With a 2N7000 and 10kohm resistor it is simple to solder an inverter on the HC06

Tygrys-1, 2021-01-19:
For module bay look at the [https://github.com/CrazyDude1994/android-taranis-smartport-telemetry](https://github.com/CrazyDude1994/android-taranis-smartport-telemetry), I am not using module bay.
For bottom port just wire 3 wires (GND, +5V and  TX on Radio side) as described above.

To make it work with the application, @CrazyDude1994 should implement protocol recognition in for multimodule.

stefaanbolle, 2021-01-19:
Thanks again!