CrazyDude1994/android-taranis-smartport-telemetry #82: Compiling

HazzaHFPV, 2020-11-21:
Not really an issue, but a question:
Could maybe anyone point me in the right direction on how I could compile the new version 1.8 myself? Unfortunately I'm a noob and have no clue how to code, build or compile apk files :/
I am willing to learn, but just need a clue on where to start.

CrazyDude1994, 2020-11-22:
It will be released at Google play soon. For compiling google "android studio". Import project there. You also need to install android SDK (I believe it will be automatically asked to be installed)

HazzaHFPV, 2020-11-24:
Cool, thank you for the pointers :) I'll install android studio and have a play-around and maybe I'll learn something along the way.

closed 2020-11-24 by HazzaHFPV

JaosnHsieh, 2021-12-09:
After put a `keystore.properties` file in the project root, I'm able to run it on android studio v4.2

keystore.properties

```
    storePassword=myStorePassword
    keyPassword=mykeyPassword
    keyAlias=myKeyAlias
    storeFile=myStoreFileLocation
```

from:

https://stackoverflow.com/a/49022618/6414615