CrazyDude1994/android-taranis-smartport-telemetry #89: Battery Info Can't View on GUI

Ansarida, 2021-01-05:
I using CRSF and using UART from taranis battery bay output by hc-05 Bluetooth.

I can get all information but the volt mah and amps now showing anything

CrazyDude1994, 2021-01-06:
can you provide your log file for this?

Ansarida, 2021-01-06:
[2021-01-06 11-57-27.zip](https://github.com/CrazyDude1994/android-taranis-smartport-telemetry/files/5775691/2021-01-06.11-57-27.zip)

Ansarida, 2021-01-07:
and also the speed not correct

Ansarida, 2021-01-14:
> can you provide your log file for this?

hi i have send the log

CrazyDude1994, 2021-01-24:
hi. Sorry I was very busy, and didn't have time to check. I will upload version 1.8.1 you may try it. It will be available here at github in the releases page, or as a BETA at google play

Ansarida, 2021-01-25:
Thank you sir.. have upload 1.8.1?

CrazyDude1994, 2021-01-25:
will upload today

CrazyDude1994, 2021-05-12:
Fixed in 1.8.4

closed 2021-05-12 by CrazyDude1994