CrazyDude1994/android-taranis-smartport-telemetry #105: Random disconnections on GalaxyS8, no full telemetry on GalaxyS8 and S7

jaroszmm, 2021-02-14:
Maybe somebody can help:
Jumper T16 with Horus Bluetooth module.
My phone sees the bluetooth as t16
I can connect with the app on my GalaxyS7 and S8.
On both phones I only get the sat number, position on the map, altitude and flight mode. No battery info (voltage or current).
On S8 I get random disconnections (sometimes after a minute, sometimes after a couple).
On S7 the connection seems to be steady.

Any ideas why it is so, what to change etc would be appreciated.

CrazyDude1994, 2021-02-21:
Hi. I don't think there is an easy to way fix this. The only solution I remeber is to manually to pair your bluetooth module to your phone first in the Android settings, and then try to connect. It solved some timeout issues for some users

jaroszmm, 2021-02-22:
OK, thanks. I think it seems to be stable on Galaxy S7, but will test it further.

mubispilvas, 2021-03-08:
The same problem, v1.8.2 could not connect to the internal Taranis q7s BLE module, 1.7.1 works fine.

jaroszmm, 2021-03-08:
So on Galaxy S7 I also get disconnections.
Meanwhile I signed for the test program (or something called like this) so I have a 'test' version from google play store of this app. I saw the current indicator working on this version but the disconnections are still present.
I guess it is not the app issue, rather the hardware on the phone.

CrazyDude1994, 2021-04-13:
You may try new version https://github.com/CrazyDude1994/android-taranis-smartport-telemetry/releases/tag/1.8.3