CrazyDude1994/android-taranis-smartport-telemetry #112: Taranis X-lite Pro with Buildin Bluetooth

druckgott, 2021-04-02:
Hallo, I have a Taranis X-lite Pro with Buildin Bluetooth. I activate the Bluetooth in the Taranis and then I can see in the App unter Bluetooth connection the "Hello" Name. I select this and it takes some seconds then I get connectain Failed. 

It do not ask me the Code "000000". What could be the reason?

druckgott, 2021-04-06:
Hallo, I found this post:
https://github.com/CrazyDude1994/android-taranis-smartport-telemetry/issues/4#issuecomment-480041038

I have a R9mini, here is the log:
[btlog.log](https://github.com/CrazyDude1994/android-taranis-smartport-telemetry/files/6266026/btlog.log)

CrazyDude1994, 2021-04-13:
Hi. You can try to use this version https://github.com/CrazyDude1994/android-taranis-smartport-telemetry/releases/tag/1.8.3

druckgott, 2021-04-14:
In this Version I have 2 versions in the menu for bluetooth in the classic I can't see the Hello
On the Lf I can see it bus if I select it the app crash down with ne error ore anything

s2kaka, 2021-04-20:
i had the same issue before , xlite pro, r9m lite pro, r9mini but now it work ok.

kusnendar, 2021-04-21:
> i had the same issue before , xlite pro, r9m lite pro, r9mini but now it work ok.
> try to Turn on location on your phone before connect app

How to pairing to android. Whats do u change name in bluetooth in your transmitter?

s2kaka, 2021-04-21:
xlite
Dont use BIG anphabet.
i just download app on google CHplay store. 
But I think problem come from ;Turn on location on  phone

druckgott, 2021-04-21:
I habe small letters and also switched on the location, but do not work the app crashed also as bevor

CrazyDude1994, 2021-04-27:
https://github.com/CrazyDude1994/android-taranis-smartport-telemetry/releases/tag/1.8.4 please try this. It still may crash, but atleast I should receive a crash report. Also after this, go to the "Settings" menu, scroll to the bottom and press "Copy debug info" then paste that info here

druckgott, 2021-04-28:
Hallo, I installed this version start the app the I press clear then I connect to Hello via LE. App crashed. I restart the app go to the settings and to copy debug, but it's empty. Is it written to a file also? So I can have a second look?

CrazyDude1994, 2021-04-28:
Do you have a Xiaomi or a Huawei device?

druckgott, 2021-04-28:
Yes it's a Huawei phone

CrazyDude1994, 2021-04-28:
I've seen crash report from your phone. I'll try to fix the issue. Also you can try to fix it yourself. Try go the application settings in Android settings, and give it a Storage permission manually, and try again

druckgott, 2021-04-28:
I already check this bevor but did not help see atacched images 
[image]

[image]

CrazyDude1994, 2021-04-29:
Try https://github.com/CrazyDude1994/android-taranis-smartport-telemetry/releases/tag/1.8.5

druckgott, 2021-04-30:
Ok now it do not crash anymore, but the connection failed

[2021.04.30 06:47] Connect to bl. Type[1] isBle[false] bondState[12] uuids[00001101-0000-1000-8000-00805f9b34fb,00000000-0000-1000-8000-00805f9b34fb,00000000-0000-1000-8000-00805f9b34fb]
[2021.04.30 06:48] Connect to bl. Type[2] isBle[true] bondState[10] uuids[null]
[2021.04.30 06:48] BLE state STATE_DISCONNECTED
[2021.04.30 06:48] BLE close connection

druckgott, 2021-04-30:
[2021.04.30 06:47] Connect to bl. Type[1] isBle[false] bondState[12] uuids[00001101-0000-1000-8000-00805f9b34fb,00000000-0000-1000-8000-00805f9b34fb,00000000-0000-1000-8000-00805f9b34fb]
[2021.04.30 06:48] Connect to bl. Type[2] isBle[true] bondState[10] uuids[null]
[2021.04.30 06:48] BLE state STATE_DISCONNECTED
[2021.04.30 06:48] BLE close connection
[2021.04.30 06:49] Connect to bl. Type[2] isBle[true] bondState[10] uuids[null]
[2021.04.30 06:49] BLE state STATE_CONNECTED
[2021.04.30 06:49] BLE data 7e 18 10 4 fffffff1 0 0 0 0 fffffffd 7e 7e 18 10 1 fffffff1 0 0 0 0 fffffff8 7e
[2021.04.30 06:49] BLE data 7e 18 10 5 fffffff1 c 0 0 0 fffffff0 7e 7e 18 10 4 fffffff1 0 0 0 0 fffffffd 7e
[2021.04.30 06:49] BLE data 7e 18 10 1 fffffff1 0 0 0 0 fffffff8 7e 7e 18 10 5 fffffff1 c 0 0 0 fffffff0 7e
[2021.04.30 06:49] BLE data 7e 18 10 4 fffffff1 0 0 0 0 fffffffd 7e 7e 18 10 1 fffffff1 0 0 0 0 fffffff8 7e
[2021.04.30 06:49] BLE data 7e 18 10 4 fffffff1 0 0 0 0 fffffffd 7e 7e 18 10 1 fffffff1 0 0 0 0 fffffff8 7e
[2021.04.30 06:49] BLE data 7e 18 10 5 fffffff1 c 0 0 0 fffffff0 7e 7e 18 10 4 fffffff1 0 0 0 0 fffffffd 7e
[2021.04.30 06:49] BLE data 7e 18 10 1 fffffff1 0 0 0 0 fffffff8 7e 7e 18 10 5 fffffff1 c 0 0 0 fffffff0 7e
[2021.04.30 06:49] BLE data 7e 18 10 4 fffffff1 0 0 0 0 fffffffd 7e 7e 18 10 1 fffffff1 0 0 0 0 fffffff8 7e
[2021.04.30 06:49] BLE data 7e 18 10 4 fffffff1 0 0 0 0 fffffffd 7e 7e 18 10 1 fffffff1 0 0 0 0 fffffff8 7e
[2021.04.30 06:49] BLE data 7e 18 10 5 fffffff1 c 0 0 0 fffffff0 7e 7e 18 10 4 fffffff1 0 0 0 0 fffffffd 7e
[2021.04.30 06:49] BLE data 7e 18 10 1 fffffff1 0 0 0 0 fffffff8 7e 7e 18 10 5 fffffff1 c 0 0 0 fffffff0 7e
[2021.04.30 06:49] BLE data 7e 18 10 4 fffffff1 0 0 0 0 fffffffd 7e 7e 18 10 1 fffffff1 0 0 0 0 fffffff8 7e
[2021.04.30 06:49] BLE data 7e 18 10 5 fffffff1 c 0 0 0 fffffff0 7e 7e 18 10 4 fffffff1 0 0 0 0 fffffffd 7e
[2021.04.30 06:49] BLE data 7e 18 10 1 fffffff1 0 0 0 0 fffffff8 7e 7e 18 10 5 fffffff1 c 0 0 0 fffffff0 7e
[2021.04.30 06:49] BLE data 7e 18 10 1 fffffff1 0 0 0 0 fffffff8 7e 7e 18 10 5 fffffff1 c 0 0 0 fffffff0 7e
[2021.04.30 06:49] BLE data 7e 18 10 4 fffffff1 0 0 0 0 fffffffd 7e 7e 18 10 1 fffffff1 0 0 0 0 fffffff8 7e
[2021.04.30 06:49] BLE data 7e 18 10 5 fffffff1 c 0 0 0 fffffff0 7e 7e 18 10 4 fffffff1 0 0 0 0 fffffffd 7e
[2021.04.30 06:49] BLE data 7e 18 10 4 fffffff1 0 0 0 0 fffffffd 7e 7e 18 10 1 fffffff1 0 0 0 0 fffffff8 7e
[2021.04.30 06:49] BLE data 7e 18 10 5 fffffff1 c 0 0 0 fffffff0 7e 7e 18 10 4 fffffff1 0 0 0 0 fffffffd 7e
[2021.04.30 06:49] BLE data 7e 18 10 5 fffffff1 c 0 0 0 fffffff0 7e 7e 18 10 4 fffffff1 0 0 0 0 fffffffd 7e
[2021.04.30 06:49] BLE data 7e 18 10 1 fffffff1 0 0 0 0 fffffff8 7e 7e 18 10 5 fffffff1 c 0 0 0 fffffff0 7e
[2021.04.30 06:49] BLE data 7e 18 10 4 fffffff1 0 0 0 0 fffffffd 7e 7e 18 10 1 fffffff1 0 0 0 0 fffffff8 7e
[2021.04.30 06:49] BLE data 7e 18 10 5 fffffff1 c 0 0 0 fffffff0 7e 7e 18 10 4 fffffff1 0 0 0 0 fffffffd 7e
[2021.04.30 06:49] BLE data 7e 18 10 1 fffffff1 0 0 0 0 fffffff8 7e 7e 18 10 5 fffffff1 c 0 0 0 fffffff0 7e
[2021.04.30 06:49] BLE data 7e 18 10 4 fffffff1 0 0 0 0 fffffffd 7e 7e 18 10 1 fffffff1 0 0 0 0 fffffff8 7e
[2021.04.30 06:49] BLE data 7e 18 10 4 fffffff1 0 0 0 0 fffffffd 7e 7e 18 10 1 fffffff1 0 0 0 0 fffffff8 7e
[2021.04.30 06:49] BLE data 7e 18 10 5 fffffff1 c 0 0 0 fffffff0 7e 7e 18 10 4 fffffff1 0 0 0 0 fffffffd 7e
[2021.04.30 06:49] BLE data 7e 18 10 4 fffffff1 0 0 0 0 fffffffd 7e 7e 18 10 1 fffffff1 0 0 0 0 fffffff8 7e
[2021.04.30 06:49] BLE data 7e 18 10 4 fffffff1 0 0 0 0 fffffffd 7e 7e 18 10 1 fffffff1 0 0 0 0 fffffff8 7e
[2021.04.30 06:49] BLE data 7e 18 10 4 fffffff1 0 0 0 0 fffffffd 7e 7e 18 10 1 fffffff1 0 0 0 0 fffffff8 7e
[2021.04.30 06:49] BLE data 7e 18 10 5 fffffff1 c 0 0 0 fffffff0 7e 7e 18 10 4 fffffff1 0 0 0 0 fffffffd 7e
[2021.04.30 06:49] BLE data 7e 18 10 1 fffffff1 0 0 0 0 fffffff8 7e 7e 18 10 5 fffffff1 c 0 0 0 fffffff0 7e
[2021.04.30 06:49] BLE data 7e 18 10 4 fffffff1 0 0 0 0 fffffffd 7e 7e 18 10 1 fffffff1 0 0 0 0 fffffff8 7e
[2021.04.30 06:49] BLE data 7e 18 10 5 fffffff1 c 0 0 0 fffffff0 7e 7e 18 10 4 fffffff1 0 0 0 0 fffffffd 7e
[2021.04.30 06:49] BLE data 7e 18 10 1 fffffff1 0 0 0 0 fffffff8 7e 7e 18 10 5 fffffff1 c 0 0 0 fffffff0 7e
[2021.04.30 06:49] BLE data 7e 18 10 1 fffffff1 0 0 0 0 fffffff8 7e 7e 18 10 5 fffffff1 c 0 0 0 fffffff0 7e
[2021.04.30 06:49] BLE data 7e 18 10 4 fffffff1 0 0 0 0 fffffffd 7e 7e 18 10 1 fffffff1 0 0 0 0 fffffff8 7e
[2021.04.30 06:49] BLE data 7e 18 10 1 fffffff1 0 0 0 0 fffffff8 7e 7e 18 10 5 fffffff1 c 0 0 0 fffffff0 7e
[2021.04.30 06:49] BLE data 7e 18 10 4 fffffff1 0 0 0 0 fffffffd 7e 7e 18 10 1 fffffff1 0 0 0 0 fffffff8 7e
[2021.04.30 06:49] BLE data 7e 18 10 5 fffffff1 c 0 0 0 fffffff0 7e 7e 18 10 4 fffffff1 0 0 0 0 fffffffd 7e
[2021.04.30 06:49] BLE data 7e 18 10 5 fffffff1 c 0 0 0 fffffff0 7e 7e 18 10 4 fffffff1 0 0 0 0 fffffffd 7e
[2021.04.30 06:49] BLE data 7e 18 10 1 fffffff1 0 0 0 0 fffffff8 7e 7e 18 10 5 fffffff1 c 0 0 0 fffffff0 7e
[2021.04.30 06:49] BLE data 7e 18 10 5 fffffff1 c 0 0 0 fffffff0 7e 7e 18 10 4 fffffff1 0 0 0 0 fffffffd 7e
[2021.04.30 06:49] BLE data 7e 18 10 1 fffffff1 0 0 0 0 fffffff8 7e 7e 18 10 5 fffffff1 c 0 0 0 fffffff0 7e
[2021.04.30 06:49] BLE data 7e 18 10 4 fffffff1 0 0 0 0 fffffffd 7e 7e 18 10 1 fffffff1 0 0 0 0 fffffff8 7e
[2021.04.30 06:49] BLE data 7e 18 10 4 fffffff1 0 0 0 0 fffffffd 7e 7e 18 10 1 fffffff1 0 0 0 0 fffffff8 7e
[... cut ...]
[2021.04.30 06:49] BLE data 7e 18 10 4 fffffff1 0 0 0 0 fffffffd 7e 7e 18 10 1 fffffff1 0 0 0 0 fffffff8 7e
[2021.04.30 06:49] BLE data 7e 18 10 5 fffffff1 c 0 0 0 fffffff0 7e 7e 18 10 4 fffffff1 0 0 0 0 fffffffd 7e
[2021.04.30 06:49] BLE data 7e 18 10 1 fffffff1 0 0 0 0 fffffff8 7e 7e 18 10 5 fffffff1 c 0 0 0 fffffff0 7e
[2021.04.30 06:49] BLE data 7e 18 10 5 fffffff1 c 0 0 0 fffffff0 7e 7e 18 10 4 fffffff1 0 0 0 0 fffffffd 7e
[2021.04.30 06:49] BLE data 7e 18 10 1 fffffff1 0 0 0 0 fffffff8 7e 7e 18 10 5 fffffff1 c 0 0 0 fffffff0 7e
[2021.04.30 06:49] No protocol detected

CrazyDude1994, 2021-04-30:
Give more info about your setup. What FC firmware do you have, which protocol do you use, your TX and RX

CrazyDude1994, 2021-04-30:
ACCESS protocol is not supported

druckgott, 2021-05-01:
Ok thx for the information will it maybe suported later?

CrazyDude1994, 2021-05-01:
Probably

s2kaka, 2021-05-03:
i use xlite pro, r9m lite pro, access.
i was crash before but now it work perfect (for about 6month)

I Name TX Bluetooth to short name "xlt", give it all permission to access store and location. turn on location and bluetooth before connect. 
I use Xiao Mi redmi 9 pro, inav, opent tx 2.3.11 ( 2.3.7 work too), 915mhz

druckgott, 2021-05-07:
How does it work with access if it's not supported? Can you post a picture von taranis how your configuration is also the reciver options you have selected?

s2kaka, 2021-05-07:
i dont know but it work, lastest access FW, r9m lite pro, f.port and s.port both work, r9mm, r9mini, r9mx. All work well. and R-xrs 2.4ghz access work too.
just normal connect.

druckgott, 2021-05-07:
ok I have tested this some minutes ago again and same probelm as bevor. have also changed name to xlt. Change access off app manually turn on bluetooth and Position and connect and then after some section get again connection failed.

s2kaka, 2021-05-07:
i have the same issue before, but after try to remove and install app some time, it work.
Try to clear data of app before remove and give it access permition to store file and access location on app setting

s2kaka, 2021-05-24:
hey, try to turn off wifi on your phone.
watch this video, he said turn on wifi will have problem. 
https://www.youtube.com/watch?v=a3wXHn0cokg.
auto translate it

druckgott, 2021-05-24:
hi thx for this, but same problem get no connection. connection failed

Enggmaug, 2021-06-22:
Hi, 

I mounted an internal bluetooth module in my RM TX16s. This is a RF-SM-02 loaded with Frsky firmware.
I installed Telemetry Viewer 1.8.6 on my android phone, and get the following log : 

`[2021.06.22 20:16] BLE state STATE_DISCONNECTED
[2021.06.22 20:16] BLE close connection
[2021.06.22 20:16] Connect to bl. Type[2] isBle[true] bondState[10] uuids[null]
[2021.06.22 20:16] BLE state STATE_CONNECTED
[2021.06.22 20:17] No protocol detected
[2021.06.22 20:17] No protocol detected
[2021.06.22 20:17] No protocol detected
[2021.06.22 20:17] No protocol detected
[2021.06.22 20:17] BLE state STATE_DISCONNECTED
[2021.06.22 20:17] BLE close connection`

If this is of any help...

CrazyDude1994, 2021-06-22:
It means that no supported protocol was detected, which means either:

- you are using unsupported protocol
- there are no data retrieved
- the data is corrupted

Enggmaug, 2021-06-23:
Thanks, 

When you say I may be using unsupported protocol, what protocol are talking of ?

I have binded RX with D16. It returns RSSI on my radio, and it displays correctly on my inav lua telemetry widget.
Even without binded RX, I would expect to see TX Batt Level.

For my test I did not plug any other sensor.

Is there a procedure to find out which one of the three possibilities is true among used protocol, data retrieved or corrupted ?

I do have oscilloscope / digital analyzer and other basic means to investigate.

Thanks for your help,
Guillaume.

Enggmaug, 2021-06-25:
Got it working. Tanks. Rssi wasn't enough...

druckgott, 2021-06-25:
How did you get it working? Can you explain what you change?

druckgott, 2021-06-25:
Ok now I also get it, didn switch on the drone l, so maybe on wrong protocol detection wite also a information in the app that you have to switch on the drone maybe I'm Stupid 😂

Enggmaug, 2021-06-25:
> How did you get it working? Can you explain what you change? 

I plugged sensors to the rx, instead of olympique relying on tx voltage ans rssi values.... and it started working.

Nandox7, 2021-11-11:
Only managed to connect and get data once with v.1.9.2
After ir disconnected and could never connect again.

From debug info I see:
```
[2021.11.10 20:35] Connect to bl. Type[2] isBle[false] bondState[12] uuids[null]
[2021.11.10 20:36] Connect to bl. Type[2] isBle[false] bondState[12] uuids[null]
[2021.11.11 11:53] Connect to bl. Type[2] isBle[false] bondState[12] uuids[null]
[2021.11.11 11:54] Connect to bl. Type[2] isBle[true] bondState[10] uuids[null]
[2021.11.11 11:54] BLE state STATE_CONNECTED
[2021.11.11 11:54] BLE data ffffffea c 14 ffffffff ffffffff 0 0 ffffffff 1 7 0 0 0 ffffff87
[2021.11.11 11:54] BLE data ffffffea c 14 ffffffff ffffffff 0 0 ffffffff 1 7 0 0 0 ffffff87
[2021.11.11 11:54] BLE data ffffffea d 3a ffffffea ffffffee 10 0 1 4 6b 0 0 3 19 fffffffa
[2021.11.11 11:54] Protocol detected e.a.a.f.a@231e518
[2021.11.11 11:55] BLE state STATE_DISCONNECTED
[2021.11.11 11:55] BLE close connection
[2021.11.11 11:55] Connect to bl. Type[2] isBle[true] bondState[12] uuids[null]
[2021.11.11 11:55] BLE state STATE_CONNECTED
[2021.11.11 11:55] No protocol detected
[2021.11.11 11:55] No protocol detected
[2021.11.11 11:56] No protocol detected
[2021.11.11 11:56] No protocol detected
[2021.11.11 11:56] BLE state STATE_DISCONNECTED
[2021.11.11 11:56] BLE close connection
[2021.11.11 11:56] Connect to bl. Type[2] isBle[true] bondState[10] uuids[null]
[2021.11.11 11:56] BLE state STATE_CONNECTED
[2021.11.11 11:56] No protocol detected
[2021.11.11 11:56] No protocol detected
[2021.11.11 11:56] BLE state STATE_DISCONNECTED
[2021.11.11 11:56] BLE close connection
[2021.11.11 11:56] No protocol detected
[2021.11.11 11:56] No protocol detected
[2021.11.11 11:57] Connect to bl. Type[2] isBle[true] bondState[10] uuids[null]
[2021.11.11 11:57] BLE state STATE_CONNECTED
[2021.11.11 11:57] No protocol detected
[2021.11.11 11:57] No protocol detected
[2021.11.11 11:57] No protocol detected
[2021.11.11 11:57] No protocol detected
[2021.11.11 11:57] Connect to bl. Type[2] isBle[true] bondState[12] uuids[null]
[2021.11.11 11:57] BLE state STATE_CONNECTED
[2021.11.11 11:57] No protocol detected
[2021.11.11 11:57] No protocol detected
[2021.11.11 11:57] No protocol detected
[2021.11.11 11:57] No protocol detected
[2021.11.11 11:58] Connect to bl. Type[2] isBle[false] bondState[12] uuids[null]
[2021.11.11 11:59] Connect to bl. Type[2] isBle[true] bondState[10] uuids[null]
[2021.11.11 11:59] BLE state STATE_CONNECTED
[2021.11.11 11:59] No protocol detected
[2021.11.11 11:59] No protocol detected
[2021.11.11 11:59] No protocol detected
[2021.11.11 11:59] No protocol detected
[2021.11.11 12:00] Connect to bl. Type[2] isBle[true] bondState[12] uuids[null]
[2021.11.11 12:00] BLE state STATE_CONNECTED
[2021.11.11 12:00] No protocol detected
[2021.11.11 12:00] No protocol detected
[2021.11.11 12:00] No protocol detected
[2021.11.11 12:00] No protocol detected
[2021.11.11 12:01] BLE state STATE_DISCONNECTED
[2021.11.11 12:01] BLE close connection
```

So it stopped detecting the protocol.