CrazyDude1994/android-taranis-smartport-telemetry #113: Crossfire problem - missing sensors + incorrect speed

edyno, 2021-04-14:
Hello everyone, thank you for this apk. However, I have a small problem with crossfire data. I'm not a programmer, but I was looking for a problem in the program.
1. The first problem is that the correct speed is not displayed. Its value is 10 times lower than reality. I read and decoded RAW data from crossfire and found that it was sent in the format speed x 10 (km/h). However, the program evaluates as follows:
             Protocol.GSPEED -> {
                 val speed = data.data / 100f
                 listener.onGSpeedData (speed)
maybe it would be enough to fix it on:
             Protocol.GSPEED -> {
                 val speed = data.data / 10f
                 listener.onGSpeedData (speed)

2. The second problem is that values such as Current, Voltage and Battery Capacity are not displayed.
The problem is possible on this line:
private const val BATTERY_PACKET_LEN = 8 (filename: CrsfProtocol.kt)
In my opinion, the value should be: BATTERY_PACKET_LEN = 9

I have no way to try it, as I don't have a compiler.
Thank you.

CapnBry, 2021-04-18:
I was just coming to report this same things!

1. The value of ground speed is in km/h / 10 in the telemetry packet. You can see it scaled here where it is (cm/s * 3.6 * 10 + 50) / 100 for a final value in km/h / 10, so edyno's code is correct it should be /10 not /100
https://github.com/betaflight/betaflight/blob/master/src/main/telemetry/crsf.c#L173

2. BATTERY_PACKET_LEN is indeed wrong for CRSF. Checking the Betaflight source it can be seen that the length sent is `CRSF_FRAME_BATTERY_SENSOR_PAYLOAD_SIZE + CRSF_FRAME_LENGTH_TYPE_CRC` or 9. All the other packet lengths are correct and include the extra 1 byte for the CRC but battery does not.
https://github.com/betaflight/betaflight/blob/master/src/main/telemetry/crsf.c#L191

avsaase, 2021-04-18:
See also https://github.com/CrazyDude1994/android-taranis-smartport-telemetry/issues/70.

CrazyDude1994, 2021-04-25:
[app-debug.zip](https://github.com/CrazyDude1994/android-taranis-smartport-telemetry/files/6373449/app-debug.zip)

try this apk. It contains your fixes. If it fixes a problem for you, I will include them in next release, meanwhile you can use this one APK for your flying sessions

CapnBry, 2021-04-26:
> [app-debug.zip](https://github.com/CrazyDude1994/android-taranis-smartport-telemetry/files/6373449/app-debug.zip)
> 
> try this apk. It contains your fixes. If it fixes a problem for you, I will include them in next release, meanwhile you can use this one APK for your flying sessions

[image]

Battery telemetry is working perfectly! Unfortunately I no longer have a working model with a GPS so I can't test the ground speed change if you've done that as well.

edyno, 2021-04-26:
Bingo !
I can confirm that everything is working correctly. Thank you :)
[image]

CrazyDude1994, 2021-04-27:
Fixed in 1.8.4

closed 2021-04-27 by CrazyDude1994

avsaase, 2021-05-01:
@CrazyDude1994 With the APK you posted here the app crashes when I click connect -> Bluetooth (Classic) -> [device name].