CrazyDude1994/android-taranis-smartport-telemetry #118: Disconnecting in Android. no visibility on ios

gdanas87, 2021-05-14:
Hello, congratulations! this app is amazing but doesn't work across all my devices. I am using the frsky ble Bluetooth module which is used for the horus, inside my jumper t16 pro (dedicated bt pads) with open tx bt option 

Lenovo tab 8 plus Android 7.0: I tried the app from the play store (1.7.1) and all other later versions from the releases repository here, and even though I can connect and see all data perfectly, I always get a disconnect about 40 seconds after connecting. 

iPhone 8 Plus ios 14.5.1, latest inav telemetry app from Apple store: At first I thought that the iPhone cant "see" the Bluetooth module inside my t16 due to the usual Apple mfi bull, but that is not the case. The inav telemetry App never sees my t16 ble bt module, but it turns out the phone can see it since I tested with speedy Bee app and to my surprise it sees the t16 (picture attached)
[image]

Samsung s10: here everything works perfectly with the "stock" play store app v1.7.1. no disconnects, very reliable. Sadly this is the only device which works flawlessly but its the device which I don't own/cant use when flying (its my wife's phone).

Now i dont have enough hope for the Lenovo tab 8 since I am sure its some hardware / Android version incompatibility issue (maybe some kind of sleep mode of the tab 8 integrated module that is buggy with ble devices??) but what about the ios app? Is there any possibility it will ever be able to "see" the ble bt devices like the one in my t16 that other apps on the phone can see (ie speedybee app)? What are my alternative options? I noticed in the ios app description the hm10 module is mentioned. Is that known to work flawlessly with the inav telemetry App on ios?

CrazyDude1994, 2021-05-14:
I know nothing about ios app. Why did you think that I made that app? My app only supports android. About the disconnet on lenovo. You are right, that's probably hardware/software on the lenovo. Few users have same issue with other brands too. One guy told me that connection stays longer if you move your transmitter further away from the drone. When it gets closer to each other you get disconnect. So you may try app at field or move drone to another room to test this

gdanas87, 2021-05-15:
> I know nothing about ios app. Why did you think that I made that app? My app only supports android. About the disconnet on lenovo. You are right, that's probably hardware/software on the lenovo. Few [...]

Thanks so much for clarifying. I thought its the same app, the fact that the ios app developer is not you was concerning, but I thought it was just some company with an ios developer account to sign your app. I will try with another android device the changes you suggest! congratulations again for this nifty little tool