CrazyDude1994/android-taranis-smartport-telemetry #145: ExpressLRS Telemetry Not Working

techfixpros, 2022-01-12:
Radio: B01 Taranis X9D
Radio FW: EdgeTx 2.5.0
Tx Module: HappyModel es900Tx
Tx Module FW: ELRS 2.0.1
Rx HappyModel es900Rx
Rx FW: ELRS 2.0.1
Tx Module: Internal non-ISRM internal XJT
Tx FW: v170317
Rx Module: BetaFPV Frsky_X SPI
Tx Module: FrSky R9M
Tx FW: ELRS 2.0.1
Quad FW: BF 2.9.11
BT Module: HC-06
Baud: 57600, 0, 1, N on internal Serial Port, w/inverter mod, w/Matek voltage regulator

Tested app with above radio and internal XJT module with the last pre-2.x FrSky firmware and I get batt voltage to the telem app on phone, but ADI does not track movement. 

Using the expresslrs setup above, I get no telem data in app at all. 

Verified in both configurations, BF Configurator shows valid stick movement.

Tested using internal serial port, and R9M S.Port.

cad435, 2022-01-25:
I might have this problem too.

I have tried both HC06 and HC05 modules. I also got the internal uninverted serial signal out if the R9M-Module (wich is flashed with ExpressLRS software). 
I confirmed it is the uninverted signal by comparing the S.Port and my "uninverted" Signal with an oscilloscope.

The modules are pairing. If I open a bluetooth serial-receiver-app on my android I indeed can see that data is transmitted to my phone. However nothing is shown in the app.

What also confuses me is the baudrate. I try to measure it and calculated it to about 80000Baud?! this would suggest that it is more like 74880Baud and not the 57600Baud suggested.

Unfortunate none of my modules can read 74880Baud , so maybe this is somethig wich should rather be talked to the ELRS-Devs?

 Can someone maybe confirm this? 

[image]

greetings 
 - cad435

howels, 2022-02-10:
I have installed the RF-BM-S02 on TX16S with EdgeTX and connected via ELRS to a quad. Telemetry working fine with defaults.

HappyModel ES24TX running eLRS 2.2.0
Matek R24D RX running eLRS 2.2.0
TX16S running EdgeTX 2.6.0 (built firmware with BLUETOOTH=ON)
RF-BM-S02 flashed as per https://www.instructables.com/Bluetooth-on-Radiomaster-TX16S/
Quad running BetaFlight 4.3.0-RC3

All I had to do was set bluetooth to "telemetry" in EdgeTX system hardware settings. Obviously for rapid telemetry updates a lower telem ratio is advisable in eLRS.

techfixpros, 2022-03-22:
> I have installed the RF-BM-S02 on TX16S with EdgeTX and connected via ELRS to a quad. Telemetry working fine with defaults.
>
> HappyModel ES24TX running eLRS 2.2.0 Matek R24D RX running eLRS 2.2.0 [...]

Telemetry, or Telemetry Mirror?

howels, 2022-03-22:
> Telemetry, or Telemetry Mirror?

The one that isn't trainer port or disabled.