CrazyDude1994/android-taranis-smartport-telemetry #155: Missing license and copyright file

Adamo198, 2023-09-20:
Hello @CrazyDude1994, 
On behalf of a Polish drone company, I would like to request the addition of a file related to the license, specifying the right to use the software. We want to use it commercially as a basis for developing their software to enhance its capabilities. Without a clear declaration regarding the license, we can't do it. I would be very happy to enter into a conversation in this regard.

CrazyDude1994, 2023-09-28:
@Adamo198 hey.
I haven't decided yet on what license it should be. But from the start I didn't want it to be used commercially. For now I'm thinking what to do in this situation.

Adamo198, 2023-09-29:
I understand, it's a difficult situation. Please think about it carefully, I don't want to put pressure on you. We are open to discussions.